rpng/open_vins #194: A problem of open_vins' function ---"StateHelper::EKFUpdate"

xhy3054, 2021-08-10:
# Hi, I found a problem of open_vins' function ---"StateHelper::EKFUpdate". 

> The function "StateHelper::EKFUpdate" of openvins is not equivalent to the traditional EKF update method. 

The whole story is: I try to integrate gnss in openvins. We know there are 4 d.o.f transformation between the ENU to VIO frames. At first, I want to test how long it takes for the vio coordinate system to converge to the ENU coordinate system if I don't estimate the transformation between ENU and VIO frames in advance. (it can be understood as using a truth value to correct a system with large error)

I tried to reproduce this problem on the KAIST dataset, First, with the "StateHelper::EKFUpdate",the result is 
[image]
We will find that it seems to show no signs of convergence(This is actually unreasonable), and i rewrite the update function(a simple EKF update process)
 ```cpp
/**
    * @brief Performs EKF update of the state 
    * @param state Pointer to state
    * @param H Jacobian of updating measurement to all variables in the system
    * @param res residual of updating measurement
    * @param R updating measurement covariance
*/
void StateHelper::EKFUpdate1(std::shared_ptr<State> state,
                            const Eigen::MatrixXd &H, const Eigen::VectorXd &res, const Eigen::MatrixXd &R) {
    const Eigen::MatrixXd P_minus = state->_Cov;
    const Eigen::MatrixXd H_trans = H.transpose();
    const Eigen::MatrixXd S = H * P_minus * H_trans + R;
    const Eigen::MatrixXd S_inv = S.llt().solve(Eigen::MatrixXd::Identity(S.rows(), S.cols()));   
    const Eigen::MatrixXd K = P_minus * H_trans * S_inv;

    const Eigen::VectorXd dx = K*res;
    
    for (size_t i = 0; i < state->_variables.size(); i++) {
        state->_variables.at(i)->update(dx.block(state->_variables.at(i)->id(), 0, state->_variables.at(i)->size(), 1));
    }

    // Update covariance.
    const Eigen::MatrixXd I_KH = Eigen::MatrixXd::Identity(state->_Cov.rows(), state->_Cov.rows()) - K * H;
    state->_Cov = I_KH * P_minus ;//* I_KH.transpose() + K * R * K.transpose();

    state->_Cov = state->_Cov.eval().selfadjointView<Eigen::Upper>();
    LimitMinDiagValue(1e-12, &state->_Cov);    
}
```
and we test on the kaist:
[image]
We find that the vio coordinate system gradually converges to the ENU coordinate system. (But using the original  "StateHelper::EKFUpdate" can not achieve this effect)

So we can draw the conclusion of that: The function "StateHelper::EKFUpdate" of openvins is not equivalent to the traditional EKF update method. We know, openvins improves its update process by means of managing covariance similar to gtsam.

(I met Mr. Huang during the interview and talked about this problem. Mr. Huang asked me to put it forward on GitHub)

(If there is something wrong with my description, I hope someone can help me point it out)

(If the above picture cannot be viewed, you can directly view it on the original web page)

yangyulin, 2021-08-10:
Hi,

I think the above code is equivalent to the original EKF update.

Can you double check the input and output for both functions and list some
examples?

By the way, for the 4 dof transformation, do you use poseJPL or self made
types?

Thanks.

xhy3054, 2021-08-10:
Hi,

The version of using "StateHelper::EKFUpdate" is
```cpp
  int size_v = 12;
  if(state->_options.do_calib_gnss_timeoffset){
  size_v ++;
  }
  Eigen::MatrixXd Hx(3, size_v);
  // J_wrt_Ta and  J_wrt_Tb is the jacobian for the pose of two adjacent frames
  Hx.block<3,6>(0,0) = J_wrt_Ta; 
  Hx.block<3,6>(0,6) = J_wrt_Tb;
  std::vector<std::shared_ptr<Type>> H_order;
  H_order.push_back(state->_clones_IMU[ta]);
  H_order.push_back(state->_clones_IMU[tb]);
  if(state->_options.do_calib_gnss_timeoffset){
  Hx.block<3,1>(0,12) = J_wrt_dt;
  H_order.push_back(state->_calib_dt_GNSStoCAM);
  }  
 gnss_queue.at(0).cov(2,2) = 1e6;

 StateHelper::EKFUpdate(state, H_order, Hx, res, gnss_queue.at(0).cov * 10);
```
The version of using original EKF update is
```cpp
  const size_t state_size = state->_Cov.rows();
  Eigen::MatrixXd Hx(3, state_size);
  Hx.setZero();
  Hx.block<3,6>(0, state->_clones_IMU[ta]->id()) = J_wrt_Ta;
  Hx.block<3,6>(0, state->_clones_IMU[tb]->id()) = J_wrt_Tb;
  if(state->_options.do_calib_gnss_timeoffset){
  Hx.block<3,1>(0, state->_calib_dt_GNSStoCAM->id()) = J_wrt_dt;
  }    
  gnss_queue.at(0).cov(2,2) = 1e6;
  StateHelper::EKFUpdate1(state, Hx, res, gnss_queue.at(0).cov *0.1);
```

I can't find the problem.

and the 4 of transformation, actually the discussion here is independent of it(it was not estimated). In the version of code where align the vio and ENU in initialization, I use the way of [cos(\theta),sin(\theta),x,y,z] to express it.

xhy3054, 2021-08-11:
This is the setting I set when debugging Gnss performance at the beginning. Unfortunately, I didn't change it in the subsequent tests. Just now I retested under the same covariance, and the results are consistent! 

Thank you very much! My previous conclusion is wrong.

closed 2021-08-11 by xhy3054

ClementLASSENA, 2022-01-07:
Hi @ xhy3054, I am interested in your integration of your GNSS measurement into Open_VINS. Is your code open source? If not, can you share any articles on this topic ?
Thank you very much !

xhy3054, 2022-01-21:
I'm sorry to reply so late, you could read the paper "Intermittent GPS-aided VIO: Online Initialization and Calibration".

xhy3054, 2022-01-21:
> 
I'm sorry to reply so late, you could read the paper "Intermittent GPS-aided VIO: Online Initialization and Calibration".

xhy3054, 2022-01-21:
> Hi @xhy3054, I am interested in your integration of your GNSS measurement into Open_VINS. Is your code open source? If not, can you share any articles on this topic ? Thank you very much !

I'm sorry to reply so late, you could read the paper "Intermittent GPS-aided VIO: Online Initialization and Calibration".