rpng/open_vins #196: Data

chenyong1003, 2021-08-19:
excuse me, I'm going to use ov_eval to convert the data format, but I'm having problems with the image.my Python and Matplotlib are already installed.
[image]

goldbattle, 2021-08-23:
You need to source the ros workspace first and the command should be
`rosrun ov_eval ov_eval <command_name>`
https://answers.ros.org/question/198279/how-to-start-a-ros-node/

You should be able to tab complete the command to ensure it is correct.

chenyong1003, 2021-08-28:
I follow your steps,but i don't know why there is such a problem in the second step.
[image]

yangyulin, 2021-08-28:
One quick solve: just go to the code of ResultSimulation and comment out
the lines with ROS_ERROR

commit 83ffb88a (2021-09-20): Fixed use or ROS function in ov_eval, fixes #196
closed 2021-09-20 by goldbattle