rpng/open_vins #199: Interesting phenomenon

hitzhang, 2021-08-23:
I didn't turn on the zupt option, but every time I was stationary, I found that the msckf point and slam point were not updated. Due to the imu integration, the pose would drift. After restarting the exercise, the pose will jump back to the position near the static state before drifting. I don't know why? There are continuous time drifting poses in the window. After retriangulation, the current frame pose is obtained. Why does it return to the pose that did not drift before?

yangyulin, 2021-08-23:
Hi,

do you have more detailed example for this?
Thanks.

hitzhang, 2021-08-24:
> Hi, do you have more detailed example for this? Thanks.
> […](#)
> -- With Best Regards, Linde Yang / Yulin Yang *UNIVERSITY of DELAWARE* PhD Candidate of Mechanical Engineering 328 Spencer Lab, Newark, DE, USA, 19716-3140 Mobile: +(01) 302-690-3926 mailto: ***@***.***> ***@***.*** Web: https://yangyulin.github.io/ <https://yangyulin.github.io/>

feature_slam ids :: 139092 139247 139289 139243 139242 139302 139294 139305 139299 139306 139300 139321 139307 139295 139366 139341 139336 139340 139372 139343 139446 139451 139462 139385 139394 139448 139449 139461 139536 139479 139464 139469 139550 139554 139547 139557 139601 139699 139641 139634 139627 139658 139773 139772 139711 139740 139774 139725 139726 139708 139705 139748 139719 139713 139729 139710 139706 139822 139823 139795 139838 139817 139843 139845 139892 139866 139891 139855 139860 139886 139862 139895 139861 139899 139927 139903 139898 139926 139924 139929 139914 139904 139934 139933 139922 139913 139918 139919 139966 139953 139980 139963 139959 139978 139983 139994 139987 140017 140023 140030 140034 140033 140028 140032 140015 140016 140014 140037 140025 140026 140035 140021 140029 140046 140057 140076 140089 140099 140100 140170 140199 140208 140262 140338 140344 140646 140696 140707 140717 140718 140719 140720 140721 140722 140723 140724 140725 140726 140727 140728 140729 140730 140731 140732 140733 140734 140735 140736 140737 140738 
pub propagator pose 1625534283.386532 -1.00254 4.04063 0.27309 0.59697 -0.53312 -0.39799 0.44834 0.26289
msckf update id: 
updaterMSCKF->update 22 0 -1.00254  4.04063  0.27309
updaterSLAM->update 0 0 -1.00254  4.04063  0.27309
bg =  | ba =  0.00393 -0.00277 0.00113 -0.23778 -0.40378 0.08331
pub esti pose 1625534283.386532 -1.003 4.041 0.273 0.597 -0.533 -0.398 0.448 0.263

feature_slam ids :: 139092 139247 139289 139243 139242 139299 139295 139307 139321 139300 139306 139305 139294 139302 139340 139343 139372 139336 139341 139366 139449 139461 139448 139394 139385 139462 139451 139446 139536 139479 139464 139469 139554 139601 139557 139547 139550 139699 139641 139634 139627 139658 139708 139706 139710 139729 139713 139719 139748 139705 139726 139725 139774 139711 139772 139773 139822 139823 139838 139860 139862 139895 139861 139866 139891 139892 139843 139855 139919 139918 139913 139922 139933 139934 139914 139926 139898 139927 139899 139929 139983 139959 139963 139980 139953 139966 139994 139987 140029 140021 140035 140034 140025 140037 140014 140015 140032 140028 140033 140016 140057 140076 140099 140170 140199 140208 140262 140338 140344 140646 140707 140721 140739 140740 140741 140742 140743 140744 140745 140746 140747 140748 140749 140750 140751 140752 140753 140754 140755 140756 140757 140758 140759 140760 140761 140762 140763 140764 140765 140766 140767 140768 140769 140770 140771 140772 140773 140774 140775 140776 
pub propagator pose 1625534283.486610 -0.97533 4.04819 0.27733 0.59642 -0.52903 -0.40221 0.45016 0.28388
msckf update id: 140017 140030 139978 139924 
updaterMSCKF->update 37 4  -1.50097  3.03971  0.03446
updaterSLAM->update 0 0 -1.50097  3.03971  0.03446
bg =  | ba =  0.00386 -0.00273 0.00123 -0.25451 -0.39941 0.07048
pub esti pose 1625534283.486610 -1.501 3.040 0.034 0.597 -0.529 -0.402 0.449 0.230

静止后 由于没有视觉更新，imu积分开始漂移，20s大概漂移30cm.

 1625534283.486610这一帧是静止后开始运动的第一帧，开始有4个msckf 点更新，但是位姿突变，从 -0.97533 4.04819 0.27733突变到 -1.50097  3.03971   0.03446，不知道为什么？

hitzhang, 2021-08-24:
[image]

hitzhang, 2021-08-24:
1625534282.98684 572.98901 542.91364 0.62636 0.59055 
1625534283.08661 573.01355 542.93378 0.62650 0.59068 
1625534283.18656 572.99280 542.91003 0.62637 0.59054 
1625534283.28676 573.18439 542.37097 0.62688 0.58798 
1625534283.38653 569.62067 531.13287 0.59948 0.52964 

可能是这几个点 三角化的几个观测大部分是静止的，导致结果误差太大，更新后对位姿影响过大？

goldbattle, 2021-09-20:
I am not sure what would cause this, but when you are still you are unable to triangulate features to constrain your pose. This is why you need the zero velocity update or a few SLAM features that can be updated.