rpng/open_vins #203: Problems during assembly on the jetyson nx

MsWik, 2021-09-16:
- Jetson xavier NX
- Jetpack 4.6
- OpenCV 4.1.1

I am trying to build this solution according to the instructions https://docs.openvins.com/index.html on a clean JetPack 4.6 image. At the stage of building I have the following problem 

Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:87 (find_package)
  open_vins/ov_core/CMakeLists.txt:7 (find_package)

Next, I build opencv from https://github.com/ros-perception/vision_opencv

cd ~/catkin_ws/src
git clone https://github.com/ros-perception/vision_opencv
cd ~/catkin_ws
catkin_make

However, I get the following problem 

CMake Error in /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find requested Boost libraries.

  Boost version: 1.65.1

  Include Boost path: /usr/include

  The following Boost libraries could not be found:

          boost_python37

Next, I build opencv from https://github.com/ros-perception/vision_opencv

cd ~/catkin_ws/src
git clone https://github.com/ros-perception/vision_opencv
cd ~/catkin_ws
catkin_make

However, I get the following problem 

CMake Error in /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find requested Boost libraries.

  Boost version: 1.65.1

  Include Boost path: /usr/include

  The following Boost libraries could not be found:

          boost_python37

Next, I change find_package(Boost REQUIRED python37) to find_package(Boost REQUIRED python) in ~/workspace/catkin_ws_ov/src/vision_opencv/cv_bridge/CMakeLists.txt and get /usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]
 String(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } } 

I tried to build OpenCV from source (4.5.0) with CUDA but got the error that it needs the ROS version of OpenCV

---
Also interested in the question is it possible to form a trajectory from the camera on the device in real time?

goldbattle, 2021-09-20:
I guess the first question is if you can build OpenVINS *without* ROS first.
It sounds like you are having trouble with your ROS install on the system.

We have personally not tried the Xavier NX, but on the TX2 and Nano if you install ROS then it should work out of the box.
If I remember correctly I followed this to get ROS:
https://www.jetsonhacks.com/2019/10/23/install-ros-on-jetson-nano/
https://github.com/JetsonHacksNano/installROS

MsWik, 2021-09-21:
Thank you for your response. So far I have not been able to build your project. Please tell me how you created open_vins on nano. I have it in stock as well.

goldbattle, 2021-09-21:
I don't have a specific guide since I do not have the nano setup right now.
- I recommend you flash the latest jetpack with OpenCV onto the unit.
- From there install ROS using the above linked scripts.
- Then can create a ROS workspace and build the project.

If you do go through this, feel free to post what you needed to do for others in the future.

goldbattle, 2021-12-06:
Ah, yes this might have been something I just addressed while I was building as I think I remember your linked github thread.
Glad you got it figured out.

closed 2021-12-06 by goldbattle