rpng/open_vins #207: question about quaternion derivative in predict_mean_rk4()

linzha0, 2021-11-22:
Hi, I am looking the propagation code and confused in quaternion derivative, hope you can help. Based on document, the quaternion derivative seems be q_dot = 0.5 * big_omega * q, but in the code it seems q_dot = 0.5 * big_omega * delta_q. May I ask why it use delta_q instead of q ? Thank you !

linzha0, 2021-11-22:
I just realized the quaternion part of rk4 is trying to get delta_q, I guess that's reason?

FengQiaojun, 2022-11-17:
I want to follow up on this question. The implementation of quaternion integration in [predict_mean_rk4()](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/Propagator.cpp#L485) is not strictly straight-forward to me either.
The related codes are here

```
Eigen::Vector4d q_0 = state->_imu->quat();

Eigen::Vector4d dq_0 = {0, 0, 0, 1};
Eigen::Vector4d q0_dot = 0.5 * Omega(w_hat) * dq_0;
Eigen::Vector4d k1_q = q0_dot * dt;

Eigen::Vector4d dq_1 = quatnorm(dq_0 + 0.5 * k1_q);
Eigen::Vector4d q1_dot = 0.5 * Omega(w_hat) * dq_1;
Eigen::Vector4d k2_q = q1_dot * dt;

Eigen::Vector4d dq_2 = quatnorm(dq_0 + 0.5 * k2_q);
Eigen::Vector4d q2_dot = 0.5 * Omega(w_hat) * dq_2;
Eigen::Vector4d k3_q = q2_dot * dt;

Eigen::Vector4d dq_3 = quatnorm(dq_0 + k3_q);
Eigen::Vector4d q3_dot = 0.5 * Omega(w_hat) * dq_3;
Eigen::Vector4d k4_q = q3_dot * dt; 
               
Eigen::Vector4d dq = quatnorm(dq_0 + (1.0 / 6.0) * k1_q + (1.0 / 3.0) * k2_q + (1.0 / 3.0) * k3_q + (1.0 / 6.0) * k4_q);
new_q = quat_multiply(dq, q_0);
```

The most reasonable explanation I can find is, we are integrating the function of error state $\tilde{q} = q \otimes \hat{q}^{-1}$. And its dynamic function is $\dot{\tilde{q}} = \frac{1}{2}\boldsymbol{\Omega}(\hat{\boldsymbol{\omega}}) \tilde{q}$. And we have the initial error $\tilde{q}(0) = [0, 0, 0, 1]$. Once we get the $\tilde{q}(\Delta t)$, we apply it on the original state to get the new estimate $\hat{q}(\Delta t) = \tilde{q}(\Delta t) \otimes \hat{q}(0)$.
Is this the right interpretation of the codes above?

yangyulin, 2022-11-18:
Yea, generally when doing rotation integration, we attempt to comput the delta rotation first. Then, combine the delta rotation with previous rotation to get the new rotation. 
 rk4 is slightly better than first order approximation

FengQiaojun, 2022-11-18:
Thank you @yangyulin !
I think I kind of get the idea of the calculating delta rotation. I think we are still using the dynamic function of $$\dot{\hat{q}} = \frac{1}{2}\boldsymbol{\Omega}(\hat{\boldsymbol{\omega}}) \tilde{q}$$ of the estimated rotation. It is not the dynamic function of the error rotation or delta rotation.
Following the RK4 method, we have
$$q_{n+1} = q_n + \left(\frac{1}{6}k_1\Delta t + \frac{1}{3}k_2\Delta t + \frac{1}{3}k_3\Delta t + \frac{1}{6}k_4\Delta t\right) = \left(1 + \left(\frac{1}{6}k_1\Delta t + \frac{1}{3}k_2\Delta t + \frac{1}{3}k_3\Delta t + \frac{1}{6}k_4\Delta t\right)q^{-1}_n\right)q_n$$
Actually I do have some questions for this. Probably for quaternion the $+$ need to be changed to $\oplus$?
I think in the code it is actually calculating 
$$dq = 1 + \left(\frac{1}{6}k_1\Delta t + \frac{1}{3}k_2\Delta t + \frac{1}{3}k_3\Delta t + \frac{1}{6}k_4\Delta t\right)q^{-1}_n$$ 
$$ki_q = k_i\Delta t q^{-1}_n \text{ for } i = 1,2,3,4$$
Every term removes the $q_n$ component so that's why we initialize 
`Eigen::Vector4d dq_0 = {0, 0, 0, 1};`. And finally times to $q_n$ on the left. (Why on the left though?)
Is this closer to the idea behind this delta rotation implementation?

goldbattle, 2022-11-23:
As Yulin says, you can think of each RK4 integration as "resetting" the quaternion to be a zero quaternion using the previous integration step. This is very different from working on the 4x1 vector, as addition of quaternions in itself is invalid.

The actual linear system can be seen here (omega dot is the system we wish the integrate in general for a quaternion):
http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf
https://docs.openvins.com/propagation.html#conti_prop
[image]

FengQiaojun, 2022-11-23:
Thank you @goldbattle !
The RK4 integration is calculating $y_{k+1}$ from $\dot{y}, y_k$. 
In this case, we have $y = \Theta$, $\dot{y} = \dot{\Theta} = \frac{1}{2}\Omega(\omega)\Theta$.
And in the code, __dq_0__ is $\Theta(t_k,t_k)$ and __dq__ is $\Theta(t_{k+1}, t_k)$. 
Then $\bar{q}(t_{k+1}) = \Theta(t_{k+1}, t_k)\bar{q}(t_{k})$, which is corresponding to the code __new_q = quat_multiply(dq, q_0);__
I guess this is the right interpretation.

goldbattle, 2022-12-22:
There is a discussion in Sola's quaterion report on integration over time too.
https://arxiv.org/pdf/1711.02508.pdf

(4.6 Time-integration of rotation rates)
[image]

(A.3 The RK4 method)
[image]

closed 2023-03-15 by goldbattle