rpng/open_vins #208: Issue about “roslaunch ov_msckf pgeneva_serial_eth.launch"

ilms123, 2021-12-02:
Hi,
   I have  a problem of roslaunch. When I run "**`roslaunch ov_msckf pgeneva_ros_eth.launch`**”, the program runs well.
But it failed when run "**`roslaunch ov_msckf pgeneva_serial_eth.launch`**".  
The error shows 
[image]

My Configuration information：ROS noetic   \ python 3.8 (the same problem using python 2.7) \  ubuntu 20.04

Thank you ! Looking forward your reply!

WoosikLee2510, 2021-12-02:
can you try running gdb (its in the launch file) and repost the error please?

ilms123, 2021-12-06:
I try to running gdb ,it shows 
[image]
But when I debug it in my another computer ,it run well. It have the same configuration information:ROS noetic \ python 3.8  \ ubuntu 20.04 
Is it the problem with system configuration? Do I need to reinstall Ros or ubuntu ?

goldbattle, 2021-12-06:
Yeah, this looks like either it is linking to a wrong version of ROS, there is some issue with your ROS install, or the bad is corrupted. You could try reinstalling ros, rebuilding from a clean workspace, and using the subscriber node. Do you only have one version of ROS installed on your system?

ilms123, 2021-12-07:
I only have one version of ROS noetic on my system, and reinstall it correctly. I rebuild a new workspace and debug it .Subscriber node runs well ,serial node shows same error . Maybe it is the problem of  compatibility of application . Weird

goldbattle, 2021-12-08:
I would recommend trying with another bag, otherwise not sure if there is anything from my side to help with.

ilms123, 2021-12-09:
I will use another bag . Thanks a lot !!!

closed 2021-12-09 by ilms123