rpng/open_vins #211: About T_imu_cam in Kaist Dataset

HarborC, 2021-12-18:
Thanks for your great application. I have a question about T_imu_cam in Kaist Dataset’s config. 

I download the dataset, and found that 

T_car_imu is
　[1.0, 0, 0, 0.07]
　[0, 1.0, 0, 0]
　[0, 0, 1.0, 1.7]
　[0,0,0,1]

and T_car_LeftCam is
　[-0.00680499, -0.0153215, 0.99985, 1.64239]
　[-0.999977, 0.000334627, -0.00680066, 0.247401]
　[-0.000230383, -0.999883, -0.015 3234, 1.58411]
　[0.0, 0.0, 0.0, 1.0]

So T_imu_LeftCam is 
　[-6.80499e-03 -1.53215e-02 9.99850e-01 1.71239e+00]
　[-9.99977e-01 3.34627e-04 -6.80066e-03 2.47401e-01]
　[-2.30383e-04 -9.99883e-01 -1.53234e-02 -1.15890e-01]
　[0,0,0,1]

But in openvins/config/kaist/ kalibr_imucam_chain.yaml, 

T_imu_LeftCam is
　[-0.00413, -0.01966, 0.99980, 1.73944]
　[-0.99993, -0.01095, -0.00435, 0.27803]
　[0.01103, -0.99975, -0.01962, -0.08785]
　[0.0, 0.0, 0.0, 1.0]

I tried to do the vio in both different config, and found yours are right. I don’t know how to caculate the T_imu_LeftCam correctly. 
Ｈope you can help me, Thanks．

goldbattle, 2021-12-18:
The script I believe I created is this one (test it out yourself):
https://github.com/rpng/open_vins/blob/master/ov_data/kaist/matlab/kaist_calculate_imu2cam.m

```matlab
clear all;
close all;

% load their stereo parameters
% looks like they used the matlab calibration tool to get it.
% https://www.mathworks.com/help/vision/ref/stereoparameters.html#d117e94134
load ../raw/stereoParams.mat
%stereoParams = toStruct(stereoParams);

% transform between the left (stereo pair base) and the right camera
% the units of the stereo parameters are millimeters
% need to get R_1to2 and p_1in2
R_2to1 = stereoParams.RotationOfCamera2;
p_2in1 = stereoParams.TranslationOfCamera2'/1000;
T_LtoR = [...
    R_2to1' p_2in1;
    0 0 0 1
];

% T_LtoR = zeros(4,4);
% T_LtoR(1,1) = 1;
% T_LtoR(1,4) = -0.4751;
% T_LtoR(2,2) = 1;
% T_LtoR(3,3) = 1;
% T_LtoR(4,4) = 1;

% Vehicle2Stereo.txt
% Stereo camera (based on left camera) extrinsic calibration parameter from vehicle
% WHAT????? THEIR POSITION IS INCORRECTLY THE OPOSITE
% DIRECTION????!@#!@#?!@#!@#$!@$%!#@%!#%!~@%!@#%@!#^%#@%^!@#%@#%
p_LinV = [1.66944; 0.278027; 1.61215];
R_LtoV = [-0.00413442 -0.0196634 0.999798; -0.999931 -0.0109505 -0.00435034; 0.0110338 -0.999747 -0.0196168];
T_VtoL = [...
    R_LtoV' -R_LtoV'*p_LinV;
    0 0 0 1
];

% Vehicle2IMU.txt
% IMU extrinsic calibration parameter from vehicle
p_IinV = [-0.07; 0; 1.7];
R_ItoV = [1 0 0; 0 1 0; 0 0 1];
T_VtoI = [...
    R_ItoV' -R_ItoV'*p_IinV;
    0 0 0 1
];

% calculate the transform between the camneras and the IMU
T_LtoI = T_VtoI*inv(T_VtoL);
T_RtoI = T_VtoI*inv(T_LtoR*T_VtoL);

% print it out
fprintf('T_C0toI = \n');
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_LtoI(1,1),T_LtoI(1,2),T_LtoI(1,3),T_LtoI(1,4));
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_LtoI(2,1),T_LtoI(2,2),T_LtoI(2,3),T_LtoI(2,4));
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_LtoI(3,1),T_LtoI(3,2),T_LtoI(3,3),T_LtoI(3,4));
fprintf('%.5f,%.5f,%.5f,%.5f\n',T_LtoI(4,1),T_LtoI(4,2),T_LtoI(4,3),T_LtoI(4,4));

fprintf('T_C1toI = \n');
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_RtoI(1,1),T_RtoI(1,2),T_RtoI(1,3),T_RtoI(1,4));
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_RtoI(2,1),T_RtoI(2,2),T_RtoI(2,3),T_RtoI(2,4));
fprintf('%.5f,%.5f,%.5f,%.5f,\n',T_RtoI(3,1),T_RtoI(3,2),T_RtoI(3,3),T_RtoI(3,4));
fprintf('%.5f,%.5f,%.5f,%.5f\n',T_RtoI(4,1),T_RtoI(4,2),T_RtoI(4,3),T_RtoI(4,4));
```

HarborC, 2021-12-19:
Thank you for your help!

closed 2021-12-19 by HarborC