rpng/open_vins #216: minor fix for two camera but not stereo type camera system

CanCanZeng, 2021-12-29:
Hi, when I want to run euroc data without enable stereo flag, the program does not run. I check the code and find that there might be a small fix for this situation.

https://github.com/rpng/open_vins/blob/057534beadca2c8d72571fed180ad289136fcdc4/ov_msckf/src/core/VioManager.cpp#L159
change to 
`size_t num_unique_cameras = (params.state_options.num_cameras == 2 && params.use_stereo) ? 1 : params.state_options.num_cameras;`

goldbattle, 2022-01-06:
Hi I was unable to reproduce this. Running on the current master (both `serial.launch` and `subscribe.launch`), and disabling `use_stereo` to false in both launch files works as expected.

```
roslaunch ov_msckf serial.launch use_stereo:=false
roslaunch ov_msckf subscribe.launch dobag:=true use_stereo:=false
```

goldbattle, 2022-01-06:
Right now if you have two cameras and they do not have sync'ed timestamps as a stereo pair the current subscribers will not work. The case you mentioned is this case. I will change the current logic to have if `use_stereo` is false, then we use monocular subscribers instead of processing the two camera system as a pair of images.

goldbattle, 2022-01-06:
Closed via d63142e642dfade328c45625ec962c2948956994

closed 2022-01-06 by goldbattle

CanCanZeng, 2022-01-06:
Yes, you are right! I feed two images in two monocular callbacks. Actually I have four cameras, so I need to use monocular callbacks since the old code does not allow feed more than 2 images at once (sorry I did not notice the recent change, there is no limit now)

CanCanZeng, 2022-01-07:
> Right now if you have two cameras and they do not have sync'ed timestamps as a stereo pair the current subscribers will not work. The case you mentioned is this case. I will change the current logic to have if `use_stereo` is false, then we use monocular subscribers instead of processing the two camera system as a pair of images.
> 
> EDIT: in this case you might need to double your clone count and we only support one timeoffset for all cameras.

I'm afraid that in my case, I do have to call monocular subscribers many times. Although I have time sync'ed cameras, their viewing directions are different, so the exposure times are different, thus the timestamps of them are not eactly the same. (I think maybe I should use the trigger time as image timestamp, but the general practice is use the  _`trigger time + 0.5 * exposure time`_  as the image timestamp).