rpng/open_vins #218: use bearing vector for feature coordinate representation instead of normalized uv

CanCanZeng, 2022-01-06:
Hi, would you please use bearing vector for feature coordinate representation instead of normalized uv? My suggestion is based on the thought that, the bearing vector is more general for all kinds of camera models, including pinhole or fisheye or omnidirectional camera.
As far as I know there are two parts need to change. One is triangulation, and the other is  mis-matches invalidation in feature tracking. The triangulate function does not need to change a lot, the bigger change happens to invalidate mis-matches in feature tracking, we cannot use cv::findFundamentalMat but need to realize a customized findEssentialMat. For this part we can refer to openvslam https://github.com/OpenVSLAM-Community/openvslam/blob/cf452c3dbc31b5a1346a14912fbdf14aa42752c7/src/openvslam/solve/essential_solver.cc#L89

I can help if you would like to do so, since I have done this to vins-mono before.

goldbattle, 2022-01-06:
I don't think this is something we want to have included currently, but feel free to make a fork with support if you want others to have this as an option. We already support pinhole and fisheye so the only reason is for > 150 degree cameras. This is already tracked in issue #184 I believe. Additionally, support for calibration and documentation of these cameras would need to be added which isn't trivial.

CanCanZeng, 2022-01-06:
You are right, my suggestion is a solution for #184 .

closed 2022-01-21 by goldbattle