rpng/open_vins #222: Running ETH3D Dataset

mhuzai, 2022-01-25:
Hello,

I have two questions regarding running open_vins with [ETH3D](https://www.eth3d.net/).

1. None of the sequences in this dataset begin with a stationary camera (to the best of my knowledge). In fact, the camera is almost never still. Would open_vins initialize and work in such a scenario? My open_vins branch is a year old, so I haven't tested with latest master, but wanted to get your feedback before updating (I'll have a lot of merge conflicts).
2. If initializing with ground truth, are the IMU biases necessary? ETH3D only provides ground truth poses and nothing else. If the biases are necessary, would there be a way to obtain them from the dataset somehow?

Thank you!

goldbattle, 2022-01-31:
Right now we require static initialization. You can try to use the groundtruth with the biases being zero, but it will depend on the quality of the velocity in the groundtruth and if that groundtruth is gravity aligned. If there is a vicon system you can try running them with our [vicon2gt](https://github.com/rpng/vicon2gt) utility to see if you can recover the full inertial states.

mhuzai, 2022-01-31:
Gotcha. Thank you for the vicon2gt idea.

One more question about initialization. Are there plans to add dynamic initialization in the future? Is there some fundamental challenge associated with doing so?

goldbattle, 2022-02-10:
Issue https://github.com/rpng/open_vins/issues/62 also has some discussion about RGBD inputs. In terms of dynamic initialization this is something we are working on and is tracked in issue https://github.com/rpng/open_vins/issues/204

mhuzai, 2022-02-14:
Great, thank you!

closed 2022-02-14 by mhuzai