rpng/open_vins #227: Trajecotry Alignment

mhuzai, 2022-02-14:
Hello,

When aligning the estimated trajectory with the ground truth trajectory, why is only position taken into account? Wouldn't taking the rotation into account help achieve a better alignment?

A related question I have is about estimating scale during alignment. Am I correct in understanding that monocular VIO cannot estimate true scale and scale has to be estimated during alignment?

Thank you!

WoosikLee2510, 2022-02-15:
1) we have multiple aligning methods, so could you be more specific about which code you are talking about?
2) Mono-vio(cam + imu) should be able to recover the scale. I think you are talking about Mono-vo (cam only)?

mhuzai, 2022-02-15:
1. I was talking about `se3`. I assumed that that was the best method to align trajectories. Is that not so? I see that `se3_single` aligns the rotation too, but at the same time it only looks at the first pose. Is there a method that both looks at the entire trajectory and does rotational alignment as well?
2. Thank you for the clarification!

goldbattle, 2022-02-15:
I recommend you take a look at this paper to clarify what exact alignment process we use. We have basically re-implemented this work exactly (which is a very nice summary):
https://rpg.ifi.uzh.ch/docs/IROS18_Zhang.pdf

The transformation we are interested in finding is aligning the two frame of references of the two trajectories, not the individual poses at each time. You might be thinking of the [hand-eye calibration](https://en.wikipedia.org/wiki/Hand_eye_calibration_problem) problem which solves a similar issue of unknown sensor frames too.

mhuzai, 2022-02-15:
I did read that before but am still confused about a few things (sorry, not a SLAM person).

If two trajectories are in the same frame of reference (e.g., both ground truth and the estimated trajectory are in the IMU's frame of reference), then aligning them means aligning the origin, correct? If so, I can see how that would involve a translation. As for rotation, is the idea that roll and pitch can be resolved but yaw can't because the gravity vector doesn't change, and so a yaw-only rotation has to be estimated too?

Another question I have is that according to Table 1 in the paper, visual-inertial requires a yaw-only rigid body transformation (4DoF). But when I use yaw-only with monocular VIO, I get worse results than `se3`. Why is that?

mhuzai, 2022-03-04:
Gotcha. What's confusing me now is that in the paper Zhang et al. used a yaw-only rigid body transformation for VINS-Mono. With monocular OpenVINS, yaw-only yields a worse result than se3. Is there some intuition as to why that may be?

goldbattle, 2022-03-07:
I would hypothesis that this is likely due to poor roll and pitch estimates.

closed 2022-03-07 by mhuzai