rpng/open_vins #229: The Jacobian of covarance propagation

RuPingCen, 2022-03-07:
Thank you for sharing this multifunctional platform. I encountered the following problems when reading the code. Adding a new camera state to the system state in MSCKF requires augmenting the covariance, such as (15)(16)in [1], but I didn't see the corresponding code segment when I read the code, can you give me some guidance?

[image]

[1] Mourikis A I, Roumeliotis S I. A Multi-State Constraint Kalman Filter for Vision-aided Inertial Navigation[C]//ICRA. 2007, 2: 6.

goldbattle, 2022-03-07:
The documentation should explain what exactly we perform, please take a look:
https://docs.openvins.com/classov__msckf_1_1StateHelper.html#ade3a86a89f15cc6e03a6430fd1dee694

> After propagation, normally we augment the state with an new clone that is at the new update timestep. This augmentation clones the **IMU pose** and adds it to our state's clone map. If we are doing time offset calibration we also make our cloning a function of the time offset. 

I hope these issues also help you:
- https://github.com/rpng/open_vins/issues/135
- https://github.com/rpng/open_vins/issues/145
- https://github.com/rpng/open_vins/issues/13

You can additionally refer to the paper state definition (eq (3)).
http://udel.edu/~pgeneva/downloads/papers/c10.pdf
[image]

RuPingCen, 2022-03-12:
Thank you for your patient guidance. I guess I have some misunderstanding about the OpenVINS . When I read the code carefully again, I have the following opinions:

1. literature [1] saves N camera poses in sliding window? So, the Jacobian matrix  **J**  of the camera state at time k+1|k relative to the IMU error state is calculated, and the system covariance **P** matrix is updated with equation (15)(16)in [1] .

2. But the sliding window of OpenVINS saves N IMU states, which are called anchor pose. So, when a new IMU state is added to the sliding window, the system covariance matrix is directly copy the variance related to the IMU. 
 
Is this statement correct? 

[1] Mourikis A I, Roumeliotis S I. A Multi-State Constraint Kalman Filter for Vision-aided Inertial Navigation[C]//ICRA. 2007, 2: 6.

WoosikLee2510, 2022-03-12:
Yes

yangyulin, 2022-03-12:
It is equivalent to understanding that the J in [1] equals the identity
matrix in OpenVINS.

closed 2022-03-14 by goldbattle