rpng/open_vins #230: Jetson Nano Developer Kit REQUIRED process [run_subscribe_msckf-1] has died!

smtsarial, 2022-03-07:
I am getting this error and log files not giving error cause. I am trying to build openvins on the jetson nano and then I will try to use them with a real-time camera. 
[image]
[master.log](https://github.com/rpng/open_vins/files/8197750/master.log)
[roslaunch-samet-11049.log](https://github.com/rpng/open_vins/files/8197751/roslaunch-samet-11049.log)
[roslaunch-samet-11268.log](https://github.com/rpng/open_vins/files/8197752/roslaunch-samet-11268.log)
[roslaunch-samet-11357.log](https://github.com/rpng/open_vins/files/8197753/roslaunch-samet-11357.log)
[roslaunch-samet-11464.log](https://github.com/rpng/open_vins/files/8197754/roslaunch-samet-11464.log)
[rosout.log](https://github.com/rpng/open_vins/files/8197755/rosout.log)
[rosout-1-stdout.log](https://github.com/rpng/open_vins/files/8197756/rosout-1-stdout.log)

goldbattle, 2022-03-07:
Are you able to run the simulation launch file? Additionally, can you try running with DEBUG verbosity and/or gdb? Can you launch with config files from the repository (for example the EurocMav)? OpenCV's YAML parser might be seg-faulting since that is just want it does on bad YAML config file formats.

What commit are you on?

smtsarial, 2022-03-07:
with the config file it is producing the same output. I am using melodic with opencv 4.1.1 can it relate to that? I am on the #226 Clang fix.

goldbattle, 2022-03-07:
Can you run with gdb and see where it is failing?
https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/subscribe.launch#L28

smtsarial, 2022-03-08:
sorry for the late I built opencv etc. here is the output of gdb 

`
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.12:39039/

SUMMARY
========

CLEAR PARAMETERS
 * /run_subscribe_msckf/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /run_subscribe_msckf/config_path: /home/samet/works...
 * /run_subscribe_msckf/max_cameras: 2
 * /run_subscribe_msckf/record_timing_filepath: /tmp/traj_timing.txt
 * /run_subscribe_msckf/record_timing_information: False
 * /run_subscribe_msckf/use_stereo: True
 * /run_subscribe_msckf/verbosity: INFO

NODES
  /
    run_subscribe_msckf (ov_msckf/run_subscribe_msckf)

ROS_MASTER_URI=http://localhost:11311

process[run_subscribe_msckf-1]: started with pid [11438]
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_          msckf...done.
Starting program: /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msc          kf __name:=run_subscribe_msckf __log:=/home/samet/.ros/log/bb19ff1e-9e7b-11ec-a904-00044b          e732e8/run_subscribe_msckf-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fad8f04a0 (LWP 11464)]
[New Thread 0x7fad0ef4a0 (LWP 11465)]
[New Thread 0x7fac8ee4a0 (LWP 11466)]
[New Thread 0x7fa7ffe4a0 (LWP 11471)]

Thread 1 "run_subscribe_m" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/aarch64/strlen.S:94
94      ../sysdeps/aarch64/strlen.S: No such file or directory.
(gdb)
`

goldbattle, 2022-03-08:
Can you run `backtrace` or `bt` in the gdb console to get were it is segfaulting.
I would ensure the config files you are opening exist.

smtsarial, 2022-03-08:
`samet@samet:~/workspace/catkin_ws_ov$ roslaunch ov_msckf subscribe.launch config:=euroc_mav
... logging to /home/samet/.ros/log/d021774a-9f28-11ec-95b5-00044be732e8/roslaunch-samet-8308.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:39905/

SUMMARY
========

CLEAR PARAMETERS
 * /run_subscribe_msckf/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /run_subscribe_msckf/config_path: /home/samet/works...
 * /run_subscribe_msckf/max_cameras: 2
 * /run_subscribe_msckf/record_timing_filepath: /tmp/traj_timing.txt
 * /run_subscribe_msckf/record_timing_information: False
 * /run_subscribe_msckf/use_stereo: True
 * /run_subscribe_msckf/verbosity: INFO

NODES
  /
    run_subscribe_msckf (ov_msckf/run_subscribe_msckf)

ROS_MASTER_URI=http://localhost:11311

process[run_subscribe_msckf-1]: started with pid [8332]
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msckf...done.
Starting program: /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msckf __name:=run_subscribe_msckf __log:=/home/samet/.ros/log/d021774a-9f28-11ec-95b5-00044be732e8/run_subscribe_msckf-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fadea94a0 (LWP 8423)]
[New Thread 0x7fad6a84a0 (LWP 8424)]
[New Thread 0x7facea74a0 (LWP 8425)]
[New Thread 0x7fa7ffe4a0 (LWP 8430)]

Thread 1 "run_subscribe_m" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/aarch64/strlen.S:94
94	../sysdeps/aarch64/strlen.S: No such file or directory.
(gdb) bt
#0  strlen () at ../sysdeps/aarch64/strlen.S:94
#1  0x0000007fb6a08f38 in _IO_vfprintf_internal (s=s@entry=0x7fffffd308, 
    format=format@entry=0x7fb7a328d0 "%s:%d: error: (%d) %s in function %s\n", 
    ap=...) at vfprintf.c:1643
#2  0x0000007fb6a9f57c in ___vsnprintf_chk (s=0x7fffffd528 "", 
    maxlen=<optimized out>, flags=1, slen=<optimized out>, 
    format=0x7fb7a328d0 "%s:%d: error: (%d) %s in function %s\n", args=...)
    at vsnprintf_chk.c:63
[... 31 lines cut ...]
    at /usr/include/c++/7/bits/shared_ptr.h:707
#20 ov_core::YamlParser::YamlParser (fail_if_not_found=true, 
    config_path="/home/samet/workspace/catkin_ws_ov/src/open_vins/ov_msckf/../config/euroc_mav/estimator_config.yaml", this=0x5555b16900)
---Type <return> to continue, or q <return> to quit---
    at /home/samet/workspace/catkin_ws_ov/src/open_vins/ov_core/src/utils/opencv_yaml_parse.h:74
#21 __gnu_cxx::new_allocator<ov_core::YamlParser>::construct<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/7/ext/new_allocator.h:136
#22 std::allocator_traits<std::allocator<ov_core::YamlParser> >::construct<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __p=<optimized out>)
    at /usr/include/c++/7/bits/alloc_traits.h:475
#23 std::_Sp_counted_ptr_inplace<ov_core::YamlParser, std::allocator<ov_core::YamlParser>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., 
    this=0x5555b168f0) at /usr/include/c++/7/bits/shared_ptr_base.h:526
#24 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<ov_core::YamlParser, std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., this=<optimized out>)
    at /usr/include/c++/7/bits/shared_ptr_base.h:637
#25 std::__shared_ptr<ov_core::YamlParser, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __tag=..., 
    this=0x7fffffe540) at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#26 std::shared_ptr<ov_core::YamlParser>::shared_ptr<std::allocator<ov_core::Yam---Type <return> to continue, or q <return> to quit---
lParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __tag=..., this=0x7fffffe540)
    at /usr/include/c++/7/bits/shared_ptr.h:344
#27 std::allocate_shared<ov_core::YamlParser, std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=...) at /usr/include/c++/7/bits/shared_ptr.h:691
#28 std::make_shared<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> ()
    at /usr/include/c++/7/bits/shared_ptr.h:707
#29 main (argc=<optimized out>, argv=<optimized out>)
    at /home/samet/workspace/catkin_ws_ov/src/open_vins/ov_msckf/src/run_subscribe_msckf.cpp:70
(gdb) 
`

WoosikLee2510, 2022-03-08:
Seems to fail while reading the configs. Can you check which one it is failing at?

smtsarial, 2022-03-08:
> 

yeah I check that one but I am not good at debugging at c. I built this project on my local ubuntu 20.04 but everything goes wrong in jetson nano. Can you give more detail about the process and the problem?

smtsarial, 2022-03-08:
After changed config part my output like that:
`samet@samet:~/workspace/catkin_ws_ov$ source devel/setup.bash 
samet@samet:~/workspace/catkin_ws_ov$ roslaunch ov_msckf subscribe.launch config:=euroc_mav
... logging to /home/samet/.ros/log/7f25ddf8-9f38-11ec-83e2-00044be732e8/roslaunch-samet-9554.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:37823/

SUMMARY
========

CLEAR PARAMETERS
 * /run_subscribe_msckf/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /run_subscribe_msckf/config_path: /home/samet/works...
 * /run_subscribe_msckf/max_cameras: 2
 * /run_subscribe_msckf/record_timing_filepath: /tmp/traj_timing.txt
 * /run_subscribe_msckf/record_timing_information: False
 * /run_subscribe_msckf/use_stereo: True
 * /run_subscribe_msckf/verbosity: INFO

NODES
  /
    run_subscribe_msckf (ov_msckf/run_subscribe_msckf)

ROS_MASTER_URI=http://localhost:11311

process[run_subscribe_msckf-1]: started with pid [9573]
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msckf...done.
Starting program: /home/samet/workspace/catkin_ws_ov/devel/lib/ov_msckf/run_subscribe_msckf __name:=run_subscribe_msckf __log:=/home/samet/.ros/log/7f25ddf8-9f38-11ec-83e2-00044be732e8/run_subscribe_msckf-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fade9f4a0 (LWP 9604)]
[New Thread 0x7fad69e4a0 (LWP 9605)]
[New Thread 0x7face9d4a0 (LWP 9606)]
[New Thread 0x7fa7ffe4a0 (LWP 9611)]

Thread 1 "run_subscribe_m" received signal SIGSEGV, Segmentation fault.
0x0000007fb7cf97f8 in boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*) ()
   from /usr/lib/aarch64-linux-gnu/libboost_filesystem.so.1.65.1
(gdb) bt
#0  0x0000007fb7cf97f8 in boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*) ()
   from /usr/lib/aarch64-linux-gnu/libboost_filesystem.so.1.65.1
#1  0x00000055555723f8 in boost::filesystem::exists (p=...)
    at /usr/local/include/boost/filesystem/operations.hpp:289
#2  ov_core::YamlParser::YamlParser (fail_if_not_found=true, 
    config_path="/home/samet/workspace/catkin_ws_ov/src/open_vins/ov_msckf/../config/euroc_mav/estimator_config.yaml", this=0x5555b20900)
    at /home/samet/workspace/catkin_ws_ov/src/open_vins/ov_core/src/utils/opencv_yaml_parse.h:68
#3  __gnu_cxx::new_allocator<ov_core::YamlParser>::construct<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/7/ext/new_allocator.h:136
#4  std::allocator_traits<std::allocator<ov_core::YamlParser> >::construct<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __p=<optimized out>)
    at /usr/include/c++/7/bits/alloc_traits.h:475
#5  std::_Sp_counted_ptr_inplace<ov_core::YamlParser, std::allocator<ov_core::YamlParser>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., 
    this=0x5555b208f0) at /usr/include/c++/7/bits/shared_ptr_base.h:526
#6  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<ov_core::Yam---Type <return> to continue, or q <return> to quit---
lParser, std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., this=<optimized out>)
    at /usr/include/c++/7/bits/shared_ptr_base.h:637
#7  std::__shared_ptr<ov_core::YamlParser, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __tag=..., 
    this=0x7fffffe6a0) at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#8  std::shared_ptr<ov_core::YamlParser>::shared_ptr<std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=..., __tag=..., this=0x7fffffe6a0)
    at /usr/include/c++/7/bits/shared_ptr.h:344
#9  std::allocate_shared<ov_core::YamlParser, std::allocator<ov_core::YamlParser>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__a=...) at /usr/include/c++/7/bits/shared_ptr.h:691
#10 std::make_shared<ov_core::YamlParser, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> ()
    at /usr/include/c++/7/bits/shared_ptr.h:707
#11 main (argc=<optimized out>, argv=<optimized out>)
    at /home/samet/workspace/catkin_ws_ov/src/open_vins/ov_msckf/src/run_subscribe_msckf.cpp:70
(gdb) 
(gdb) 
**`
My output for dpkg -l | grep libopencv**
[image]

goldbattle, 2022-03-10:
It looks like you might have conflicting opencv versions. Can you try to manually change the cmakelists to point to the 3.2 version? or re-install your 4.1 with the contribs? I think the jetson already have opencv, thus you shouldn't need to install it yourself.

smtsarial, 2022-03-12:
I solved this issue with opencv 4.1.1 and cuda integration. I started to write a guide for jetson nano which will cover dataset tests and real-time monocular cases with an Inertial measurement unit. wish me luck

goldbattle, 2022-03-14:
Great, feel free to link here if you make it public.

closed 2022-03-14 by goldbattle