rpng/open_vins #233: How do I do OpenVINS without a dataset(rosbag file)?

dirux0101, 2022-03-19:
I want to run OpenVINS directly in realtime using realsense D435 camera. How do I change the code?

goldbattle, 2022-03-19:
First you would need to make a configuration file for that camera + IMU setup.
https://docs.openvins.com/gs-calibration.html
https://github.com/rpng/open_vins/tree/master/config

Then you would run the driver for the sensor, and launch the subscribe node which will subscribe to the live camera (see tutorial for more details if needed).
https://docs.openvins.com/gs-tutorial.html

dirux0101, 2022-03-20:
> First you would need to make a configuration file for that camera + IMU setup. https://docs.openvins.com/gs-calibration.html https://github.com/rpng/open_vins/tree/master/config
>
> Then you would run [...]

Thank you for your answer. Can you be a little more specific?
And I wonder if it's possible to just connect the camera to the laptop, not the drone.

goldbattle, 2022-03-20:
We require an IMU sensor. You would need to provide another sensor for this. We use ROS, which you would need to use the realsense driver to publish images onto (along with whatever IMU you are using).
http://wiki.ros.org/realsense2_camera

There is no way to directly connect to a camera unless you code / implement that yourself.

dirux0101, 2022-03-21:
> 

OK. Thank you so much. It helped a lot. I'll try that information you gave me!

closed 2022-03-21 by dirux0101

xym0601, 2025-05-12:
> 我们需要一个 IMU 传感器。您需要为此提供另一个传感器。我们使用 ROS，您需要使用 RealSense 驱动程序将图像发布到 ROS 上（以及您正在使用的任何 IMU）。http [://wiki.ros.org/realsense2_camera](http://wiki.ros.org/realsense2_camera)
> 
> 除非您自己编写代码/实现，否则无法直接连接到相机。

If it is running in docker, is it also published on ros?