rpng/open_vins #235: build error on ubuntu16.04 ROS1

CanCanZeng, 2022-03-24:
Hi, when I build the latest code 5cf101423d4f5b26ef2ac6ff03eeefd59401f661 on my desktop with ubuntu16.04 ROS1 environment, there comes some error:
```
zcc@zcc-desk:~/WS_ROS/open_vins_v2.6$ catkin build
--------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/zcc/WS_ROS/open_vins_v2.6
--------------------------------------------------------------------
Build Space:        [exists] /home/zcc/WS_ROS/open_vins_v2.6/build
Devel Space:        [exists] /home/zcc/WS_ROS/open_vins_v2.6/devel
Install Space:      [unused] /home/zcc/WS_ROS/open_vins_v2.6/install
Log Space:         [missing] /home/zcc/WS_ROS/open_vins_v2.6/logs
Source Space:       [exists] /home/zcc/WS_ROS/open_vins_v2.6/src
DESTDIR:            [unused] None
--------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/bin/catkin", line 11, in <module>
    load_entry_point('catkin-tools==0.6.1', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 272, in main
    catkin_main(sysargs)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 267, in catkin_main
    sys.exit(args.main(args) or 0)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/cli.py", line 422, in main
    summarize_build=opts.summarize  # Can be True, False, or None
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 283, in build_isolated_workspace
    workspace_packages = find_packages(context.source_space_abs, exclude_subspaces=True, warnings=[])
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 87, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 148, in find_packages_allowing_duplicates
    xml, filename=filename, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 771, in parse_package_string
    raise InvalidPackage('Error(s):%s' % (''.join(['\n- %s' % e for e in errors])), filename)
catkin_pkg.package.InvalidPackage: Error(s) in package '/home/zcc/WS_ROS/open_vins_v2.6/src/open_vins/ov_msckf/package.xml':
Error(s):
- The generic dependency on 'std_msgs' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'geometry_msgs' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'sensor_msgs' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'nav_msgs' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'cv_bridge' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'image_transport' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'ov_core' is redundant with: build_depend, build_export_depend, exec_depend
- The generic dependency on 'ov_init' is redundant with: build_depend, build_export_depend, exec_depend
```

After comment out many dependencies my compilation success finally, and OV runs well. I don't know why this should come.
I have tested on two ubuntu16.04 desktops, they have the same problem.

goldbattle, 2022-03-24:
It looks like it is a to old of version of `catkin_pkg` maybe? See this issue: https://github.com/ros-infrastructure/catkin_pkg/issues/303#issuecomment-768464197

Can you look at that thread and report your catkin version?
```
pip list | grep catkin
```

The [docker 16.04](https://github.com/rpng/open_vins/blob/master/Dockerfile_ros1_16_04) build which runs on [xenial](https://github.com/osrf/docker_images/blob/master/ros/kinetic/ubuntu/xenial/desktop-full/Dockerfile#L3) is fine, so maybe try updating your packages?
https://github.com/rpng/open_vins/runs/5651935634?check_suite_focus=true

CanCanZeng, 2022-03-25:
the output of `pip list | grep catkin` is 
```
catkin                        0.7.20                 
catkin-pkg                    0.4.22                 
catkin-pkg-modules            0.4.22                 
catkin-tools                  0.6.1
```

CanCanZeng, 2022-03-25:
after I upgrade pip and reinstall catkin_pkg, the output of `pip list | grep catkin` becomes
```
catkin                        0.7.20
catkin-pkg                    0.4.24
```
but the redundant dependency problem still exists.

goldbattle, 2022-03-25:
Can you also run
```
apt-cache policy python-catkin-pkg
dpkg -L python-catkin-pkg
```

Can you look around and make sure this version is above `4.23` https://github.com/ros-infrastructure/catkin_pkg/issues/303#issuecomment-768464197

CanCanZeng, 2022-03-26:
```
apt-cache policy python-catkin-pkg
python-catkin-pkg:
  Installed: 0.4.22-100
  Candidate: 0.4.22-100
  Version table:
 *** 0.4.22-100 100
        100 /var/lib/dpkg/status
     0.2.10-2 500
        500 http://mirrors.huaweicloud.com/repository/ubuntu xenial/universe amd64 Packages
        500 http://mirrors.huaweicloud.com/repository/ubuntu xenial/universe i386 Packages
```

```
dpkg -L python-catkin-pkg
/.
/usr
/usr/bin
/usr/bin/catkin_create_pkg
/usr/bin/catkin_find_pkg
/usr/bin/catkin_generate_changelog
/usr/bin/catkin_package_version
/usr/bin/catkin_prepare_release
/usr/bin/catkin_tag_changelog
/usr/bin/catkin_test_changelog
/usr/lib
/usr/lib/python2.7
/usr/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info/PKG-INFO
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info/dependency_links.txt
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info/entry_points.txt
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info/requires.txt
/usr/lib/python2.7/dist-packages/catkin_pkg-0.4.22.egg-info/top_level.txt
/usr/share
/usr/share/doc
/usr/share/doc/python-catkin-pkg
/usr/share/doc/python-catkin-pkg/changelog.Debian.gz
/usr/share/doc/python-catkin-pkg/copyright
```

I find that if I install catkin_pkg by `sudo python -m pip install catkin_pkg`, the latest version 0.4.24 is installed under `/usr/local/` but my python executable is under `/usr/bin/` , if I run `sudo apt install python-catkin-pkg --upgrade` , the package is installed under `/usr/lib` and the version is 0.4.22

Finally I find that if I add `export PYTHONPATH=/usr/local/lib/python2.7/site-packages:/usr/lib/python2.7/site-packages:$PYTHONPATH` , the error is gone!

closed 2022-03-27 by CanCanZeng