rpng/open_vins #236: Error while building using Eigen 3.4.0

yhabib29, 2022-04-07:
There is an error while building open_vins using Eigen 3.4.0:
```
/home/user/open_vins/catkin_ws/src/open_vins/ov_msckf/src/ros/ROS1Visualizer.cpp:709:129:   required from here
/usr/include/eigen3/Eigen/src/Core/Dot.h:74:3: error: static assertion failed: YOU_TRIED_CALLING_A_VECTOR_METHOD_ON_A_MATRIX
   74 |   EIGEN_STATIC_ASSERT_VECTOR_ONLY(Derived)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Core/Dot.h:75:3: error: static assertion failed: YOU_TRIED_CALLING_A_VECTOR_METHOD_ON_A_MATRIX
   75 |   EIGEN_STATIC_ASSERT_VECTOR_ONLY(OtherDerived)
```
[image]

Eigen linked Issue: [#1664](https://gitlab.com/libeigen/eigen/-/issues/1664)

Solution:
Replace line **709** in **`open_vins/ov_msckf/src/ros/ROS1Visualizer.cpp`** by :
```diff
- double ori_nees = 2 * quat_diff.block(0, 0, 3, 1).dot(covariance.block(0, 0, 3, 3).inverse() * 2 * quat_diff.block(0, 0, 3, 1));
+ double ori_nees = 2 * quat_diff.block<3,1>(0, 0).dot(covariance.block<3,3>(0, 0).inverse() * 2 * quat_diff.block<3,1>(0, 0));

```

goldbattle, 2022-04-07:
Wow, I always thought the two methods were the same, but it seems not. Thanks for the detailed investigation.

goldbattle, 2022-04-07:
It looks like this was fixed in the ROS2 but not the ROS1 in https://github.com/rpng/open_vins/pull/226

Can you try the following?
```
  Eigen::Vector3d quat_diff_vec = quat_diff.block(0, 0, 3, 1);
  Eigen::Vector3d cov_vec = covariance.block(0, 0, 3, 3).inverse() * 2 * quat_diff.block(0, 0, 3, 1);
  double ori_nees = 2 * quat_diff_vec.dot(cov_vec);
  Eigen::Vector3d errpos = state_ekf.block(4, 0, 3, 1) - state_gt.block(5, 0, 3, 1);
  double pos_nees = errpos.transpose() * covariance.block(3, 3, 3, 3).inverse() * errpos;
```

yhabib29, 2022-04-07:
Yes, I just tried your PR and it builds without problems !
Thanks

commit 0de470d3 (2022-04-07): fix eigen asserts for ROS1 viz (closes #236)
closed 2022-04-07 by goldbattle