rpng/open_vins #238: Theory about zero velocity update

CanCanZeng, 2022-04-09:
Hi, where can I find the theory about the zero velocity update in your code? The publication does not mention it.
And why do you use `&&` instead of `||` for zero velocity detection? https://github.com/rpng/open_vins/blob/d84a51c2b914b5739ed6eaff30eb74e7d6a687a3/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L192
It sounds more reasonable to me that it is considered static only if both conditions are met at the same time. Please correct me if I'm wrong.

goldbattle, 2022-04-12:
There is a small discussion on the documentation website:
https://docs.openvins.com/update-zerovelocity.html

You can enforce that both are there, but the IMU detection is highly dependent on the quality of the IMU. For example on the KAIST dataset, just using the IMU detection will work well, but on other datasets it won't without the user spending quite a bit of time tuning.

CanCanZeng, 2022-04-14:
Thank you for your help, I got it.

closed 2022-04-14 by CanCanZeng