rpng/open_vins #239: Running OpenVINS without datasets(rosbag)

dirux0101, 2022-04-09:
Hello

I calibrated RealSense D455 using Kalibr and got imu_chain.yaml and imucam_chain.yaml files.

Now I need to create a config file and start the Subscribe.launch file? Is that right?
Is there any part that needs to be modified in the launch file?

I'm not familiar with this part, so I'd appreciate it if you could explain in detail.
Thank you

goldbattle, 2022-04-12:
Yes. I recommend you follow the getting started tutorial to get an understanding on how to use the configuration and launch files. It also has a more basic launch file without a lot of the logging you can use as a starting point. If you are still confused after reading the description / tutorial there, feel free to ask for clarification.
https://docs.openvins.com/gs-tutorial.html

dirux0101, 2022-05-11:
> 네. 구성 및 시작 파일을 사용하는 방법에 대한 이해를 얻으려면 시작하기 자습서를 따르는 것이 좋습니다. 또한 시작점으로 사용할 수 있는 로깅이 많지 않은 보다 기본적인 시작 파일이 있습니다. 거기에 있는 설명/튜토리얼을 읽은 후에도 여전히 혼란스럽다면 자유롭게 설명을 요청하십시오. [https://docs.openvins.com/gs-tutorial.html ](https://docs.openvins.com/gs-tutorial.html)

I'm sorry I saw this post late.

By referring to the basic configuration of the launch file in the tutorial, we even created the launch file. The configuration file was imported from euroc_mav, but I got an error saying the configuration file could not be opened.

I couldn't find a basic configuration example for a configuration file in the tutorial. Is there a configuration file for real time? I'm sorry if I didn't see the example even though it's in the tutorial.

Thank you for your help.

goldbattle, 2022-05-11:
All config files are in the config folder.

dirux0101, 2022-05-18:
> 모든 구성 파일은 구성 폴더에 있습니다.
> [… ](#)

I tried copying the estimator_config.yaml file from 'euroc_mav' or another file, but an error appears and the process ends. Can you tell me how to solve it?

goldbattle, 2022-05-18:
I recommend you look in the launch file to see how the arguments are being used.
If you moved the config, then you need to specify the whole `bag` and `config_path` paths.
https://github.com/rpng/open_vins/blob/master/ov_msckf/launch/subscribe.launch#L30-L44

If you still have trouble after understanding this, feel free to reopen the issue.

dirux0101, 2022-05-18:
I already tried it by replacing the "config_path" part of the launch file with the name of the newly created config. I also tried referring to the launch file in the tutorial, but it didn't work. Should I comment out the bag file part in the subscribe.launch file?

goldbattle, 2022-05-18:
Can you post the error? Please provide enough information for debugging,
otherwise it is hard to help.

goldbattle, 2022-05-18:
> When I roslaunch the subscribe.launch file, the error is as follows.
> [image: screen]
> <https://user-images.githubusercontent.com/101799943/169062915-db4b21e7-6994-4382-8b09-4a54139afab4.png>
>
> Here is the config folder I created myself.
> [image: screen1]
> <https://user-images.githubusercontent.com/101799943/169063049-f9d91be5-4789-4d30-9da0-dfeab0d6e097.png>

Can you zip and upload these config files?
Additionally, what is the command you are running to launch this, please
provide this.

dirux0101, 2022-05-18:
1. roscore
2. roslaunch realsense2_camera rs_camera.launch filters:=pointcloud unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true 
3. roslaunch ov_msckf subscribe.launch

Here is the config folder file I created.
[custom.zip](https://github.com/rpng/open_vins/files/8717155/custom.zip)

Here is the modified subscribe.launch file and the custom.launch file i created.
[launch.zip](https://github.com/rpng/open_vins/files/8717174/launch.zip)

goldbattle, 2022-05-18:
Your configs are not valid, add this to the top of each
```
%YAML:1.0 # need to specify the file type at the top!
```

dirux0101, 2022-05-18:
Thanks. I tried adding it to the each config file I created, but another error appeared.
[image]

dirux0101, 2022-05-18:
I solved it by editing the header file. The problem went on to show nothing in rviz,.,

goldbattle, 2022-05-18:
What you screenshotted is not an error, it is just letting you know it will
find the other transform.
You will need to play the bag (as explained in the online tutorial) or
specify "dobag" to be true in the launch file if you specified the bag path.

dirux0101, 2022-05-18:
I tried playing with the dataset. I want to do openvins with my camera in real time with 'dobag' set to false and without bag file. Is it possible?

yunkong11, 2022-10-21:
Hi, I want to ask you why I play my own data package, but openvins cannot run. Where do I modify and add the topic？thankyou

goldbattle, 2022-10-31:
As mentioned above, you need to create the config files for the camera and IMU with the correct topics. From there you can use the existing launch files. See the example config files in the `config/` folder.

lwttttt, 2024-03-30:
hi? how could I run the openvins without rosbag