rpng/open_vins #240: Running open_vins with gazebo and px4

stevedanomodolor, 2022-04-14:
``System 
ros2 foxy 
Problem: 
I am trying to run the open_vis to estimate VIO for a simulation I made with the iris drone with a front camera. Given that the message type is not sensor_msg::msg but px4_msgs::sensorCombined, I modified the code in the ros2visualiser, specifically the imu subscirber. This is the modification I made. 

```
void ROS2Visualizer::callback_inertial(const px4_msgs::msg::SensorCombined::UniquePtr msg) {

  // convert into correct format
  ov_core::ImuData message;
  // message.timestamp = msg->header.stamp.sec + msg->header.stamp.nanosec * 1e-9;
  // message.wm << msg->angular_velocity.x, msg->angular_velocity.y, msg->angular_velocity.z;
  // message.am << msg->linear_acceleration.x, msg->linear_acceleration.y, msg->linear_acceleration.z;
  //
  message.timestamp = msg->timestamp/1000000.0;
  message.wm << msg->gyro_rad[0], msg->gyro_rad[1], msg->gyro_rad[2];
  message.am << msg->accelerometer_m_s2[0], msg->accelerometer_m_s2[1], msg->accelerometer_m_s2[2];

```

Eveytime i run the simulation I get this error 
```
1649928195.6166582 [INFO] [launch]: All log files can be found below /home/stevedan/.ros/log/2022-04-14-11-23-15-615892-stevedan-OMEN-Laptop-15-ek0xxx-209021
1649928195.6167920 [INFO] [launch]: Default logging verbosity is set to INFO
1649928195.7044764 [INFO] [run_subscribe_msckf-1]: process started with pid [209023]
1649928195.7046418 [INFO] [rviz2-2]: process started with pid [209025]
1649928195.8140919 [run_subscribe_msckf-1] initialized imu conv node
1649928195.8167100 [run_subscribe_msckf-1] [32moverriding node [1m[32mverbosity[0m[32m with value from ROS!
1649928195.8168349 [run_subscribe_msckf-1] [0mSetting printing level to: INFO
1649928195.8334203 [run_subscribe_msckf-1] [32moverriding node [1m[32mmax_cameras[0m[32m with value from ROS!
1649928195.8335309 [run_subscribe_msckf-1] [0m[32moverriding node [1m[32mmax_cameras[0m[32m with value from ROS!
1649928195.8335917 [run_subscribe_msckf-1] [0m[32moverriding node [1m[32muse_stereo[0m[32m with value from ROS!
1649928195.8336411 [run_subscribe_msckf-1] [0m[32moverriding node [1m[32mmax_cameras[0m[32m with value from ROS!
1649928195.8336775 [run_subscribe_msckf-1] [0m[32moverriding node [1m[32muse_stereo[0m[32m with value from ROS!
1649928195.8337104 [run_subscribe_msckf-1] [0mLogging tets
1649928198.3971076 [run_subscribe_msckf-1] [34m[TIME]: 0.0030 seconds total (337.8 hz)
1649928198.3972430 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0040 seconds total (249.9 hz)
1649928198.3973083 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0024 seconds total (414.1 hz)
1649928198.3973594 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0015 seconds total (678.0 hz)
1649928198.3974059 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0015 seconds total (684.0 hz)
1649928198.3974519 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0020 seconds total (494.1 hz)
1649928198.3974969 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0016 seconds total (613.5 hz)
1649928198.3975430 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0014 seconds total (730.5 hz)
1649928198.3975887 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0015 seconds total (662.7 hz)
1649928198.3976338 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0018 seconds total (559.9 hz)
1649928198.3976791 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0039 seconds total (259.1 hz)
1649928198.3977242 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0038 seconds total (265.6 hz)
1649928198.3977687 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0031 seconds total (318.2 hz)
1649928198.3978133 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (389.7 hz)
1649928198.3978581 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0057 seconds total (176.8 hz)
1649928198.3979025 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0042 seconds total (240.7 hz)
1649928198.3979473 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0044 seconds total (229.0 hz)
1649928198.3979921 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (381.7 hz)
1649928198.3980362 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0025 seconds total (401.8 hz)
1649928198.3980803 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (385.5 hz)
1649928198.3981245 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0029 seconds total (340.7 hz)
1649928198.3981690 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (384.2 hz)
1649928198.3982134 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0024 seconds total (421.4 hz)
1649928198.3982575 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0024 seconds total (412.4 hz)
1649928198.3983023 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (382.8 hz)
1649928198.3983469 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (383.7 hz)
1649928198.3983982 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (388.2 hz)
1649928198.3984427 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0025 seconds total (399.4 hz)
1649928198.3984876 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0037 seconds total (272.7 hz)
1649928198.3985324 [run_subscribe_msckf-1] [0m[34m[TIME]: 0.0026 seconds total (388.7 hz)
1649928198.3985775 [run_subscribe_msckf-1] [0m[32m[init]: successful initialization in 0.0003 seconds
1649928198.3986351 [run_subscribe_msckf-1] [0m[32m[init]: orientation = 1.0000, 0.0000, -0.0009, 0.0057
1649928198.3986821 [run_subscribe_msckf-1] [0m[32m[init]: bias gyro = -0.0009, -0.0054, -0.0001
1649928198.3987265 [run_subscribe_msckf-1] [0m[32m[init]: velocity = 0.0000, 0.0000, 0.0000
1649928198.3987715 [run_subscribe_msckf-1] [0m[32m[init]: bias accel = 0.0001, -0.0006, 0.0559
[... 231 lines cut ...]
    - [0.0, 1.0, 0.0, 0.0]
    - [0.0, 0.0, 1.0, 0.0]
    - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [1]
  camera_model: pinhole
  distortion_coeffs: [0,0,0,0,0]
  distortion_model: plumb_bob
  intrinsics: [554.3827128226441, 554.3827128226441, 320.5, 240.5] #fu, fv, cu, cv
  resolution: [640, 480]
  rostopic: /front_camera/image_raw
# cam1:
#   T_imu_cam: #rotation from camera to IMU, position of IMU in camera
#     - [0.0125552670891, -0.999755099723, 0.0182237714554, -0.0198435579556]
#     - [0.999598781151, 0.0130119051815, 0.0251588363115, 0.0453689425024]
#     - [-0.0253898008918, 0.0179005838253, 0.999517347078, 0.00786212447038]
#     - [0.0, 0.0, 0.0, 1.0]
#   cam_overlaps: [0]
#   camera_model: pinhole
#   distortion_coeffs: [-0.28368365,0.07451284,-0.00010473,-3.55590700e-05]
#   distortion_model: radtan
#   intrinsics: [457.587, 456.134, 379.999, 255.238] #fu, fv, cu, cv
#   resolution: [752, 480]
#   rostopic: /cam1/image_raw

```

goldbattle, 2022-04-14:
From your printout, it looks like there is an issue with the camera and IMU timestamps. I suggest you take a look at those and see if they are reasonable if you plot them (relative to the start time). It looks like the IMU is off by `1649928076.969715` seconds from the camera times. Looking at the PX4 message I think getting the timestamp is a bit more complicated:
- https://github.com/PX4/px4_msgs/blob/master/msg/SensorCombined.msg
- https://github.com/PX4/PX4-Autopilot/issues/13968
- https://github.com/PX4/px4_msgs/blob/master/msg/VehicleImu.msg#L3-L4

```
uint64 timestamp # time since system start (microseconds)
int32 RELATIVE_TIMESTAMP_INVALID = 2147483647 # (0x7fffffff) If one of the relative timestamps is set to this value, it means the associated sensor values are invalid
```

You might want to try this [VehicleImu](https://github.com/PX4/px4_msgs/blob/master/msg/VehicleImu.msg#L3-L4) which seems to have the sample timestamp which should be the raw time? Not sure, as I have never directly interfaced with PX4 sensor data.
```
uint64 timestamp # time since system start (microseconds)
uint64 timestamp_sample # the timestamp of the raw data (microseconds)?
```

Another option you can try is to just use the current ROS timestamp if you are running it on an actual robot. This means that both the camera and IMU are all using ROS time, and OpenVINS would just need to calibrate the delay between PX4 sending and the node receiving the IMU reading. This solution wouldn't work for rosbag playback through...

stevedanomodolor, 2022-04-15:
I plot the two timestamp relattve to its starting point and I get this plot.  Now I am using the timestamp_sample from the vehicleimu, this plot makes sense right?
[image]

goldbattle, 2022-04-15:
From the axis it looks like there is still a 2 second difference? Is this `time_camera_relative` what is on the camera image topic headers that you are subscribing to?

stevedanomodolor, 2022-04-15:
The relative time is the time from when I subscribed to the topic. Time_xxxx_relative = current timestamp -xxxx_suscribe_start_time

goldbattle, 2022-04-18:
You can try writing a small node that just takes the px4 custom message, and then re-published a `sensor_msgs/IMU` topic. You could then just directly timestamp it with `ros::Time::now()` (or whatever the ros2 version of this is) in this node, and see how that performs. OpenVINS would then subscribe to this new topic and the existing camera topics you have.

stevedanomodolor, 2022-04-26:
I was not able to fix, so I added a new imu to the simulation and used that instead.

closed 2022-04-26 by stevedanomodolor