rpng/open_vins #246: Running the TUM-VI dataset in stereo mode has large angular errors during evaluation

successkaisadadi, 2022-05-07:
When I used run_ros_tumvi.sh to run the TUM-VI dataset, I found that the angle error reached 3-5deg. In other papers citing OpenVINS, the angle error is less than 2deg. Is there a similar solution?

goldbattle, 2022-05-08:
I have not done an evaluation in a while. Will try to take a look as soon as I get time, thanks for opening an issue.

successkaisadadi, 2022-05-08:
Thanks for your reply, do you have any suggestions for now or what I need to try to solve this problem?

goldbattle, 2022-06-22:
What OpenCV version are you using?

goldbattle, 2022-07-11:
It looks like this is an issue with accidently deleting IMU readings. Not sure how why I used the original logic of the oldest feature take. If there is a drop in tracking (as in the beginning of the TUM datasets), all IMU readings will be deleted / dropped. I will push a fix ASAP but you can change these lines:
https://github.com/rpng/open_vins/blob/bf0ada9f4f6d671f472616ab00e5df3bd6d6ce83/ov_msckf/src/core/VioManager.h#L80-L85

To the following:
```cpp
  // The oldest time we need IMU with is the last clone
  // We shouldn't really need the whole window, but if we go backwards in time we will
  double oldest_time = state->margtimestep();
  if (oldest_time > state->_timestamp) {
    oldest_time = -1;
  }
```

Let me know if this fixes the issue for you.

closed 2022-07-18 by goldbattle

goldbattle, 2022-10-11:
I am not sure, as I have not looked into the problem and there are not many
details.
I guess one large change is that we have added dynamic initialization in
v2.6, which in the past wouldn't have been used.

PR #259 (2022-07-18): Develop v2.6.1
- Bug fix accidental deletion of IMU when time tracking was lost (fixes #246)
- Fix ROS2 param reading in visualizer (fixes #249)
- Fix bug that the first ever IMU reading is not past into the VioManger
- Dynamic initialization is now disabled by default on most datasets 
- Now use predeclared classes and move headers to cpp files to improve compile time
- Rename ov_init simulator to `SimulatorInit` to prevent name conflicts
- Faster feature extraction logic by only extracting in sub-grids which need features (150hz for mono, 60hz for stereo)
- New simplified serial reader which [... cut]