rpng/open_vins #249: save_total_state parameter: ParameterAlreadyDeclaredException using subscribe.launch.py

pglira, 2022-05-11:
When running
```
ros2 launch ov_msckf subscribe.launch.py config:=euroc_mav
```
from the tutorial https://docs.openvins.com/gs-tutorial.html I get the following error:
```
[run_subscribe_msckf-1] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
[run_subscribe_msckf-1]   what():  parameter 'save_total_state' has already been declared
[ERROR] [run_subscribe_msckf-1]: process has died [pid 11507, exit code -6, cmd '/ros_ws/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ --params-file /tmp/launch_params_qa5bw91m --params-file /tmp/launch_params_0mltn1c9 --params-file /tmp/launch_params_0q29k68h --params-file /tmp/launch_params_t72tj74y --params-file /tmp/launch_params_effor9qu'].
```
Commenting out this line
https://github.com/rpng/open_vins/blob/bf0ada9f4f6d671f472616ab00e5df3bd6d6ce83/ov_msckf/launch/subscribe.launch.py#L87
"solves" the issue.

hodnajit, 2022-06-14:
Hi, I am facing the same error. May the error comes from this line
run_subscribe_msckf.cpp line 64
`options.automatically_declare_parameters_from_overrides(true);`

hodnajit, 2022-06-14:
I am facing the same error while running it on galactic docker ov_ros2_20_04 :)

goldbattle, 2022-06-14:
You should be able to change these lines:
https://github.com/rpng/open_vins/blob/0923323a4ed10a3f82554af938157b54ad941d4b/ov_msckf/src/ros/ROS2Visualizer.cpp#L87-L89

To something like this:
```cpp
if (node->has_parameter("save_total_state")) {
  node->get_parameter<T>("save_total_state", save_total_state);
}
```

PR #259 (2022-07-18): Develop v2.6.1
- Bug fix accidental deletion of IMU when time tracking was lost (fixes #246)
- Fix ROS2 param reading in visualizer (fixes #249)
- Fix bug that the first ever IMU reading is not past into the VioManger
- Dynamic initialization is now disabled by default on most datasets 
- Now use predeclared classes and move headers to cpp files to improve compile time
- Rename ov_init simulator to `SimulatorInit` to prevent name conflicts
- Faster feature extraction logic by only extracting in sub-grids which need features (150hz for mono, 60hz for stereo)
- New simplified serial reader which [... cut]
commit 1d4cf858 (2022-07-18): fix getting sim save param in ros2 (closes #249)
closed 2022-07-18 by goldbattle