rpng/open_vins #253: OpenVINS  estimator_config.yaml parameters explanation

vswdigitall, 2022-06-12:
Hi,
 
I found your OpenVINS project and want to try it for SLAM on moving car in city environment.

I tried it on EUROC indoor datasets and it works.

My setup: InvenSense MPU-9250, ONSemi AR0234(1280x720, 60fps. global shutter), RPI4 64bit OS, ROS Noetic.

I made IMU-camera calibration with Kalibr and recorded bag file (100sec video in my room, /imu0 and /cam0/image_raw topics).

Have launched VINS with roscore and rviz.

After few seconds INIT stage completed. Now i can see /cam0/image with keypoints and /imu0 vector. Image attached.

Would you be so kind to help me with next questions?

1. Could you explain me [estimator_config.yaml](https://github.com/rpng/open_vins/blob/master/config/euroc_mav/estimator_config.yaml) parameters for my IMU mono SLAM setup?

2. How to use VINS topics, where i can see SLAM track, pointcloud and other computed data?

3. What IMU and camera you can advice for outdoor MonoCamera IMU SLAM ( camera resolution, fps, global/rolling, IMU rate, scales)?

Best regards, Viktor.

vswdigitall, 2022-06-22:
I got some results with Arducam 1mp stereo OV9281 on Jetson NX JP 4.4 ROS Melodic and mpu9250.
Now i will install more stable IMU like XSens.

[Track](https://cam360.tv/etc/track.jpg)

Next i want to apply GPS data to VINS track. How i can do it? 
What software you can advice me?

Best regards, Viktor.

goldbattle, 2022-06-22:
How are you finding the noise parameters for the IMU?
You can follow this guide to find them:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#from-the-allan-standard-deviation-ad

This paper might be of interest:
https://pgeneva.com/downloads/papers/Lee2020ICRA.pdf

vswdigitall, 2022-06-22:
I made kalibr with https://github.com/gaowenliang/imu_utils

goldbattle, 2022-07-11:
Yes you can integrate a GPS, but you would need to implement that yourself. The current code is just visual+inertial.

vswdigitall, 2022-07-12:
Ok.

goldbattle, 2022-07-15:
Feel free to reopen / make another issue if you have more questions.
I have tried to comment the eurocmav config a bit more, hope this helps.
https://github.com/rpng/open_vins/blob/develop_v2.6.1/config/euroc_mav/estimator_config.yaml

closed 2022-07-15 by goldbattle