rpng/open_vins #256: 'libGL error: MESA-LOADER: failed to retrieve device information' can't open gazebo gui window in building docker option

tarafelmi, 2022-06-17:
getting error for the command that should open gazebo GUI on your main desktop window- 

docker run -it --net=host --gpus all \
    --env="NVIDIA_DRIVER_CAPABILITIES=all" \
    --env="DISPLAY" \
    --env="QT_X11_NO_MITSHM=1" \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    osrf/ros:noetic-desktop-full \
    bash -it -c "roslaunch gazebo_ros empty_world.launch"

get error following 

... logging to /root/.ros/log/95dfa344-ee39-11ec-9ccd-70cd0d56e19a/roslaunch-pc-ThinkPad-P1-Gen-4i-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pc-ThinkPad-P1-Gen-4i:36235/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [34]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 95dfa344-ee39-11ec-9ccd-70cd0d56e19a
process[rosout-1]: started with pid [44]
started core service [/rosout]
process[gazebo-2]: started with pid [47]
process[gazebo_gui-3]: started with pid [51]
[ INFO] [1655469164.322874787]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1655469164.323640791]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
libGL error: MESA-LOADER: failed to retrieve device information
[ INFO] [1655469164.370562637]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1655469164.371324382]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
Segmentation fault (core dumped)
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[gazebo-2] process has died [pid 47, exit code 139, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/root/.ros/log/95dfa344-ee39-11ec-9ccd-70cd0d56e19a/gazebo-2.log].
log file: /root/.ros/log/95dfa344-ee39-11ec-9ccd-70cd0d56e19a/gazebo-2*.log
Aborted (core dumped)
[gazebo_gui-3] process has died [pid 51, exit code 134, cmd /opt/ros/noetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/root/.ros/log/95dfa344-ee39-11ec-9ccd-70cd0d56e19a/gazebo_gui-3.log].
log file: /root/.ros/log/95dfa344-ee39-11ec-9ccd-70cd0d56e19a/gazebo_gui-3*.log
MESA-LOADER: failed to retrieve device information

anyone knows how to solve this?

goldbattle, 2022-06-20:
I recommend googling about running docker to try to debug. This is likely an issue with passing your GPU through to the docker container. Hope this helps.

closed 2022-07-11 by goldbattle

goldbattle, 2022-10-11:
You might want to google since this is an issue with the ros docker / demo.
Maybe your graphics card setup isn't right with docker or something, not
sure.