rpng/open_vins #257: Use visual disparity to determine whether to add a frame into the sliding window?

carolo44, 2022-06-28:
This function may improve the performance in the large-scale outdoor environment applications.

goldbattle, 2022-07-11:
Dynamic cloning is a very interesting idea and there are a few works which have looked at it a bit. The main problem is how to handle feature processing when you don't have clone states. This isn't something we look to support in this repo, but feel free to fork and play with yourself!

closed 2022-07-11 by goldbattle

carolo44, 2022-09-16:
> Dynamic cloning is a very interesting idea and there are a few works which have looked at it a bit. The main problem is how to handle feature processing when you don't have clone states. This isn't something we look to support in this repo, but feel free to fork and play with yourself!

Here is my implementation.
if the latest frame is considered as not having enough disparity with respect to its previous frame, then its feature points are simply dropped. Then the IMU measurements get propagated but no state gets cloned. Only key-frames get cloned. This Implementation handles the large-scale outdoor scenarios quite well so far(deployed on a fixed-wing drone).