rpng/open_vins #269: Failed to override value of launch file using yaml, monocular does not work

hoangvietdo, 2022-08-20:
Hi, thank you for sharing your work.

I git clone, built, and ran the project for a custom stereo camera dataset and it worked well.
However when I tried to change the YAML file to the monocular camera (i.e., ```use_stereo: false, max_cameras: 1```), the simulation still showed up as 2 cameras.

To run the mono simulation, I directly modify the ```subscriber.launch```, even though it runs correctly in this case but the algorithm is stuck at the initial process and no pose estimation is performed.

goldbattle, 2022-08-20:
The launch file always has priority. Either don't override in the launch so
it uses the one in the yaml, or edit in the launch file. Should be a print
out in console if something gets overwritten.

hoangvietdo, 2022-08-21:
@goldbattle Thanks for replying.
In the first place, I did not touch the launch at all, I only tried to modify the yaml, but the simulation has no response on that. Probably I should directly modify the launch file in the future.
Can you comment on the monocular simulation? It does not work for me.

closed 2022-09-06 by goldbattle