rpng/open_vins #270: libboost_filesystem.a(operations.o): relocation R_X86_64_PC32 against symbol

mahmood-ul-hasan, 2022-08-24:
When I am running command catkin_make it produce following error:

/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_filesystem.a(operations.o): relocation R_X86_64_PC32 against symbol `_ZN5boost6system6detail10cat_holderIvE24system_category_instanceE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [open_vins/ov_init/CMakeFiles/ov_init_lib.dir/build.make:485: /home/aisl/crane_catkin/devel/lib/libov_init_lib.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:2338: open_vins/ov_init/CMakeFiles/ov_init_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Would you please help me to solve this issue?

goldbattle, 2022-09-06:
Your installed boost might not match the one compiled with. I recommend re-installing libboost-dev and go from there.
You might want to do some googling and see if this thread helps you: https://stackoverflow.com/a/23187221

goldbattle, 2022-09-19:
Feel free reopen if still an issue.

closed 2022-09-19 by goldbattle