rpng/open_vins #273: measurement noise should _noises.sigma_w_2 * dt * dt in Zupt ?

chennuo0125-HIT, 2022-09-06:
when I see the code **UpdaterZeroVelocity.cpp**, and I found the follow measurement noise may be false, noise should be proportional to dt^2 ?  
https://github.com/rpng/open_vins/blob/cd866b5ba58f1b59f0d6983c0a379753d25d1308/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L165

goldbattle, 2022-09-06:
It should depend on the definition of Q.
https://github.com/rpng/open_vins/blob/cd866b5ba58f1b59f0d6983c0a379753d25d1308/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L178-L182

I will try to find some time to more closely remind myself of this code, and see if you are correct.

chennuo0125-HIT, 2022-09-06:
@goldbattle 
the R measure noise matrix will be used in EKFUpdate(), should proportion to dt ?
https://github.com/rpng/open_vins/blob/cd866b5ba58f1b59f0d6983c0a379753d25d1308/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L260

goldbattle, 2022-11-22:
I believe this should be correct. The noises that we have are continuous time, thus we need to convert them into a discrete time one. This is described in [Trawny's report](http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf):
[image]

The random walks are related by a sqrt(dt) while the white noises are related by a 1/sqrt(dt). When you find the uncertainty of this measurement, you get sigma^2 which equates to a times dt for random walks, and a 1/dt for white noise terms.

This definition here actually is incorrect and should use the random walk noises squared, which I will push a fix to the master in a bit:
https://github.com/rpng/open_vins/blob/cd866b5ba58f1b59f0d6983c0a379753d25d1308/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L178-L182

Let me know if this helps.

closed 2022-12-06 by goldbattle