rpng/open_vins #274: There is something wrong with compiling in `Debug` mode

zzhh00, 2022-09-06:
When I change the `Release` to `Debug` mode in the CMakeLists.txt, the error occurs as below :

`In file included from /usr/include/c++/9/cassert:44,
                 from /usr/include/eigen3/Eigen/Core:261,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/hz/hz/SLAM/Visual-Slam/Filter-based-slam/MSCKF/ov_ws/src/open_vins/ov_core/src/utils/sensor_data.h:25,
                 from /home/hz/hz/SLAM/Visual-Slam/Filter-based-slam/MSCKF/ov_ws/src/open_vins/ov_msckf/src/state/Propagator.h:31,
                 from /home/hz/hz/SLAM/Visual-Slam/Filter-based-slam/MSCKF/ov_ws/src/open_vins/ov_msckf/src/state/Propagator.cpp:22:
/usr/include/assert.h:81:13: error: declaration of ‘void __assert(const char*, const char*, int) throw ()’ has a different exception specifier
   81 | extern void __assert (const char *__assertion, const char *__file, int __line)
      |             ^~~~~~~~
`
Can you please help me solve this problem? Thank you very much!

goldbattle, 2022-09-06:
Can you try the PR here:
https://github.com/rpng/open_vins/pull/272

goldbattle, 2022-09-09:
Feel free to reopen if this is still an issue.

closed 2022-09-09 by goldbattle