rpng/open_vins #278: Suddenly failing after good performance

raabuchanan, 2022-10-06:
Hello,
I'm running OpenVINS on a pretty easy log of a legged robot walking around a courtyard outside. OpenVINS works just fine until a specific moment where suddenly most of the visual features disappear and then the state diverges. I've added a picture of Rviz of just before the state diverges. 

My best guess is that most of the good features are very far away and maybe there is some parameter setting a maximum landmark distance from the camera? Any ideas? Thanks.

Right before divergence
[image]

goldbattle, 2022-10-09:
Do you have the bag? How many features are tracked during this, or is it due to the state diverging? Can you provide your calibration / launch / data?

goldbattle, 2022-11-21:
Feel free to reopen if you have a specific bag to reproduce this, thanks!

closed 2022-11-21 by goldbattle