rpng/open_vins #282: openvins cann't run with my own dataset

yunkong11, 2022-10-22:
Hi, openvins cann't run when I play my dataset, I have read the tutorial but I don't konw the reason, I guess the position of topic is wrong. Can you teach me how to modify the configuration file of my dataset or do you have a template?

yunkong11, 2022-10-22:
Sorry for the late reply because of the file transfer problem，here is my config file，please help me see where is the problem, thank you.
[realsense_d435i.zip](https://github.com/rpng/open_vins/files/9843390/realsense_d435i.zip)

goldbattle, 2022-10-22:
You likely need to change `rostopic: /cam0/image_raw` in
kalibr_imucam_chain.yaml and `rostopic: /imu0` to your IMU in
kalibr_imu_chain.yaml.

yunkong11, 2022-10-22:
Sorry to bother you，I didn't check the topic carefully. But there is another problem: when I run my own dataset, after walking a circle, I deliberately let it stand still, and the track drifted greatly. Openvins has ZUPT, but why does it drift so huge?
[image]

yunkong11, 2022-10-22:
Hi, I find that ZUPT dosen't happen when the d435i is still after walking for a circle. Have you ever encountered such a situation? 
I think the IMU in d435i is so noisy  that it drift.
Or it is not calibrated well，but It can track well before still，so I think this is not the main reason.
I also run the MH_01_easy.bag, MH_2_easy.bag，the config parameter use both your and VINS-Mono, the terminal prompts that ZUPT occurs，but  they all drift when static.  **_This does not happen with binocular inertial_**
And you know why? Why? Look forward to your reply.

goldbattle, 2022-11-21:
I don't have any experience with the d435i, but if you search other issues, many have had problems with its IMU. I am not sure why the zero-velocity is failing, but this is something that still need to investigate (see issue https://github.com/rpng/open_vins/issues/273)

goldbattle, 2022-11-22:
Can you please try the latest master, a small bug was fixed in the bias noise model: https://github.com/rpng/open_vins/commit/38610105652a7bc9843f3fb35c7b93a528725680

closed 2022-12-01 by goldbattle