rpng/open_vins #286: Hi

jiachenglihxl, 2022-11-18:


jiachenglihxl, 2022-11-18:
I have read the dynamic initialization part, and, I am wandering if it is possible to only use linear result to init the VIO. Because solve the whole BA requires a lot Computing Power

WoosikLee2510, 2022-11-18:
What's 'linear result'?

yangyulin, 2022-11-18:
Linear results are the one directly got by IMU and SFM alignment. 

The linear results can be used for initialization. But they are too noisy. Maybe will have a higher failure rates. 

Initialization is a one-time thing. I feel it is worth to wait for a good initialization for the system to start

jiachenglihxl, 2022-11-21:
> 
Thanks for your reply, but the thing is that, for example, to solve the full BA may requires more than 0.6s in my Raspberry Pi(less than 30 iterations). If we need to calculate the covariance, it may cost even more... This means after we get the initial state, we need to integrate more than 0.6s to the current which may cause error. 
By the way, what if I use the stereo, and I get the accurate extrinsic parameter， do you think, is it still necessary to solve the full BA? I think in this configuration using the Linear results may enough.

goldbattle, 2022-11-21:
You can try to skip the optimization problem, but I would recommend just tuning the BA parameters it to reduce its run time. You can reduce the number of threads (better for embedded platforms), and reduce the number of features and number of keyframes used.

The issue with just using the linear system is how to recover the initial covariance of the system? This isn't that straightforward, but one could simply hardcode an initial covariance, and directly use the linear system result (estimator might become over confident, or diverge possibly).

closed 2022-12-01 by goldbattle

jiachenglihxl, 2022-12-03:
> You can try to skip the optimization problem, but I would recommend just tuning the BA parameters it to reduce its run time. You can reduce the number of threads (better for embedded platforms), and [...]

Thanks for you reply, for the covariance issue, I find in the static initial part, the covariance is a just hardcoded one. So, is the covariance really that matters?

By the way , I find in my Raspberry Pi, sloving the initial covariance is really slow and easily to fail. I still haven't figure out the reason.

goldbattle, 2022-12-06:
When the system is static, one can very easily design the covariance prior. When moving, how does one determine how accurate the initial orientation and velocity is?