rpng/open_vins #288: Error loading module matplotlib.pyplot?

bobododosjl, 2022-11-19:


goldbattle, 2022-11-19:
You need to make sure you have matplotlib in your path and installed on
your system.

bobododosjl, 2022-11-21:
I already check that , it satisify the requirements, still meet the same problem。

bobododosjl, 2022-11-21:
> You need to make sure you have matplotlib in your path and installed on your system.
> […](#)
I already check that , it satisify the requirements, still meet the same problem.

goldbattle, 2022-11-21:
Can you see if this issue helps you: https://github.com/lava/matplotlib-cpp/issues/8

Try running the following to see if you run into any errors.
```
python3
import matplotlib.pyplot 
```

bobododosjl, 2022-11-22:
> mport matplotlib.pyplot

is ok, i run the command  rosrun ov_eval plot_trajectories posyaw /home/bobododo/GNC/MSF/workspace4/catkin_ws_ov/src/open_vins/ov_eval/example/stamped_groundtruth.txt /home/bobododo/GNC/MSF/workspace4/catkin_ws_ov/src/open_vins/ov_eval/example/stamped_traj_estimate.txt
 still meet the  following problem:

[COMP]: 36382 poses in stamped_groundtruth => length of 80.63 meters
[COMP]: 3638 poses in stamped_traj_estimate => length of 79.92 meters
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error loading module matplotlib.pyplot!
[1]    8442 abort (core dumped)  rosrun ov_eval plot_trajectories posyaw

goldbattle, 2022-11-22:
Run in your terminal and see if you can import matplotlib yourself:
```
python3
import matplotlib.pyplot 
```

goldbattle, 2022-12-01:
Feel free to reopen if the above commands ran successfully, and you still had the problem.

closed 2022-12-01 by goldbattle