rpng/open_vins #290: Visual Thermal Inetial Dataset could be appled to this module?

MobilityAI, 2022-11-23:


goldbattle, 2022-11-23:
Thermal cameras are very different from visual cameras. You would need to either change the tracking to use the full 10bit information, or scale the incoming thermal images to the 0-255 intensity based images that could then be used with traditional feature tracking methods.

viciopoli01, 2022-11-24:
@MobilityAI please give a look at this for thermal cameras: https://github.com/jpl-x/x_multi_agent

goldbattle, 2022-12-01:
> Online Photometric Calibration of Automatic Gain Thermal Infrared Cameras
> Manash Pratim Das, Larry Matthies, Shreyansh Daftry
https://arxiv.org/abs/2012.14292
https://github.com/mpdmanash/thermal_photometric_calibration
https://www.youtube.com/watch?v=364V3xBG1Tg

closed 2022-12-01 by goldbattle