rpng/open_vins #291: Localization hardware setup

TalFeiner, 2022-11-24:
Hi,

First of all this is a great project. Thank you for sharing :)
I am trying to create a robust localization setup for a ground robot.
My current setup includes intel nuc, cheap IMU and three cameras of my choice (I can have as many cameras as I like), each camera points to a different direction.

It will be much appreciated, if you could recommend a certain setup that is likely to work well and be robust.
Would you recommend using stereo cameras over the mono cameras?
Do you think that Fisheye cameras would yield better results?
When driving a wobbly surface, the IMU gather measurements-noise.
How would you suggest to deal with such noise?
Does this project support multiple sets of stereo cameras?
Last question, if you had to create the best localization hardware setup for a ground robot, what would it be?

Thank you for your time.

goldbattle, 2022-11-25:
Stereo cameras allow for much more stable state estimation, but at a higher computational cost. Fisheye cameras are very very beneficial for state estimation. I have not had much experience with very high vibration environments, but you can try simply inflating the inertial measurement noises that are found during static / non-vibration calibration.

The project currently does not support more than 2 cameras.

Typically, for a ground robot, the calibration is the hard part, so if you can have a small detachable sensor rig that you can calibrate in 6DoF motion using Kalibr, and then fix the calibration online, this will give you the best result. There are issues with performing online calibration for ground vehicles, so I recommend avoiding that.
https://ieeexplore.ieee.org/abstract/document/8616792

TalFeiner, 2022-11-27:
> Stereo cameras allow for much more stable state estimation, but at a higher computational cost. Fisheye cameras are very very beneficial for state estimation. I have not had much experience with [...]

Thanks

closed 2022-11-27 by TalFeiner