rpng/open_vins #292: Run openvins with live camera

ergys00, 2022-11-30:
I want to run openvins with my Realsense T265 camera but I'm having trouble understanding how it all runs. 
Can I skip writing the .yaml configuration files and simply make the .launch file such as it can read the imu_info, camera_info topics and use the information published in those topics for the configuration?
 
Is the configuration file a must? Can i run openvins without the configuration files? 
When i try to do that it gives me an error saying that it can't read the configuration file

WoosikLee2510, 2022-11-30:
It's probably because your path to the config file is wrong. I recommend double-checking your path in the .launch file.
If your path is correct but still seeing the problem, you can try passing `false` arg to the `parse_config` function (e.g., [link](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/core/VioManagerOptions.h#L385-L386)). 
If this doesn't work, you can replace all the `parser->parse_config` with `(ros_node_handler)->param()` which directly accesses the ros parameters you set in your launch file.

goldbattle, 2022-12-01:
The recommendation is to use the config file system. Please see the examples in the repo:
https://github.com/rpng/open_vins/tree/master/config

The camera and IMU are a nearly identical format to that of [kalibr](https://github.com/ethz-asl/kalibr).

Otherwise, you will need to override all parameters in your launch file directly, and likely need to disable the code to try to open a config file (or point it to an empty yaml file).

ergys00, 2022-12-01:
I've set configuration files with the right values that i got from the tool rs-enumerat-devices fo my realsense t265 but when i try to launch subscribe.launch of ov_msckf the path diverges, initially it stays stable but after some movements it starts to diverge

closed 2022-12-29 by goldbattle

goldbattle, 2023-01-05:
I have uploaded an example configs here: https://github.com/rpng/open_vins/tree/master/config/rs_t265