rpng/open_vins #293: VIOManager does not initialize when running with EuRoC.

hungdche, 2022-12-01:
HI. So I was trying to run open_vins with the EuRoC dataset and grab the state as it is being generated. Here is the rough code of it
``` 
ov_core::ImuData imu_datum = {imu_timestamp, angular_v.cast<double>(), linear_a.cast<double>()};
open_vins_estimator.feed_measurement_imu(imu_datum);

cv::Mat white_mask(752, 480, CV_8UC3, cv::Scalar(255,255,255));
ov_core::CameraData cam_datum;
cam_datum.timestamp = cam_timestamp;
cam_datum.sensor_ids.push_back(0);
cam_datum.sensor_ids.push_back(1);
cam_datum.images.push_back(cam_0);
cam_datum.images.push_back(cam_1);
cam_datum.masks.push_back(white_mask);
cam_datum.masks.push_back(white_mask);
open_vins_estimator.feed_measurement_camera(cam_datum);

state = open_vins_estimator.get_state(); // does not get updated
```
The config file is the default one provided by open_vins (in `config/euroc_mav/estimator_config.yaml`). I would really appreciate any help. Thanks!

goldbattle, 2022-12-01:
I recommend you refer to the [ros1_serial_msckf.cpp](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/ros1_serial_msckf.cpp) and how data is passed and processed.
What is reported to console when you run your code?
You can change the log level with this in your main file (assuming it isn't over written someplace else):
```cpp
ov_core::Printer::setPrintLevel("ALL");
```

I believe your mask is also incorrect see what we set as the default:
https://github.com/rpng/open_vins/blob/8600ceeadce43297192d328653f016254e4594b5/ov_msckf/src/core/VioManagerOptions.h#L283
https://github.com/rpng/open_vins/blob/8600ceeadce43297192d328653f016254e4594b5/ov_msckf/src/ros/ROS1Visualizer.cpp#L458-L464

hungdche, 2022-12-01:
This is what printed out to the console: 
[image]

For the masks, I want the entire camera image, so I set `use_mask` to false. And I load in the white images as masks just in case.

hungdche, 2022-12-02:
The error is indeed because of the mask. Thanks for helping!

closed 2022-12-02 by hungdche