rpng/open_vins #296: DynamicInitializer : 538 line

jiaozhadexiaolang, 2022-12-10:
p_FinI0 = x_hat.block(size_feature * A_index_features.at(feat.first) + 6, 0, 3, 1);

jiaozhadexiaolang, 2022-12-10:
A_index_features.at(feat.first) + 6

jiaozhadexiaolang, 2022-12-10:
size_feature * A_index_features.at(feat.first) + 6
should 
A_index_features.at(feat.first)

why?  +6

jiaozhadexiaolang, 2022-12-10:
v and g in the end of x_hat

goldbattle, 2022-12-29:
The state should be the features, then velocity, then gravity
https://github.com/rpng/open_vins/blob/c3259430f0574c2911ff8e6ce888f6a1fcff5928/ov_init/src/dynamic/DynamicInitializer.cpp#L224-L231
https://github.com/rpng/open_vins/blob/c3259430f0574c2911ff8e6ce888f6a1fcff5928/ov_init/src/dynamic/DynamicInitializer.cpp#L368-L369

closed 2022-12-29 by goldbattle

qsisi, 2023-05-28:
Sorry to bother you again, but in my understanding, the x_hat is ordered as feat、vel、grav:

https://github.com/rpng/open_vins/blob/58afb2daf47498911100edc7452ccd7e42a2db50/ov_init/src/dynamic/DynamicInitializer.cpp#L477-L478

but when accessing the features:

https://github.com/rpng/open_vins/blob/58afb2daf47498911100edc7452ccd7e42a2db50/ov_init/src/dynamic/DynamicInitializer.cpp#L538

there is a "+6" at indexing, does it wrong?

commit 2d5ed9ac (2023-06-08): fix #296 more comments and switch to sparse qr recovery for speed