rpng/open_vins #298: masks in tum_vi

fratopa, 2022-12-28:
I am trying to run Open Vins using the tum_vi configuration using the command:
ros2 launch ov_msckf subscribe.launch.py config:=tum_vi
having set the verbosity setting to DEBUG I am presented with the following error:

run_subscribe_msckf-1] VioManagerOptions.h:279 VioManager(): invalid mask path:
[run_subscribe_msckf-1] VioManagerOptions.h:280         - mask0 - /home/francesco/workspace/catkin_ws_ov/install/ov_msckf/share/ov_msckf/config/tum_vi/../../ov_data/masks/tumvi0.png
[ERROR] [run_subscribe_msckf-1]: process has died

I solved the problem by copying the two .png files into the configuration folder and leaving the path empty.

It seems quite an inelegant solution, is there a better one?

goldbattle, 2022-12-29:
The config file masks are specified relative to the location of the mask files. I am not sure how best to have this besides just place the two masks next to the config files and specify `tumvi0.png` as the path. When building the contents of the config directory are copied to the install folder, so the masks would be copied over.

You can try this and see if it fixes the issue for you.

fratopa, 2022-12-29:
I agree, copying the two masks indeed solves the problem. However, by default, it seems that the masks should stay in their original location. I believe either there is a problem with the relative path creation since `/home/francesco/workspace/catkin_ws_ov/install/ov_msckf/share/ov_msckf/config/tum_vi/../../ov_data/masks/tumvi0.png` is not a valid path, or it should be added a comment specifying that the masks should be copied over in the config file.

goldbattle, 2023-01-05:
I moved them in https://github.com/rpng/open_vins/commit/d83ce4fd5b8c384ca12cf0ab567f4c044536d8cf

closed 2023-01-05 by goldbattle