rpng/open_vins #300: Kaist dataset

donghuhu, 2023-01-04:
I see that the extrinsic parameters of the camera in the "config" folder are slightly different from the ones posted by the author of the kaist dataset. I wonder which parameter is better. Thank you.

goldbattle, 2023-01-04:
The transforms in the datasets are in the vehicle frame, while we need the camera to IMU transform.
You can see how they were calculated in this matlab script:
https://github.com/rpng/open_vins/blob/master/ov_data/kaist/matlab/kaist_calculate_imu2cam.m

Hope this helps.

donghuhu, 2023-01-05:
Thank you for your help. 
But I found that the transform matrix from the left camera to vehicle posted by the author of the kaist dataset is R: -0.00680499 -0.0153215 0.99985 -0.999977 0.000334627 -0.00680066 -0.000230383 -0.999883 -0.0153234 and T: 1.64239 0.247401 1.58411. It is a little diffient from the config used in this project. Would you mind me asking which parameter is better.
[Vehicle2Stereo.txt](https://github.com/rpng/open_vins/files/10348563/Vehicle2Stereo.txt)

goldbattle, 2023-01-05:
It seems that the config is pulled from the `sample_with_img.tar.gz` dataset which seems to be different from all the other datasets. I will push the correct config files in https://github.com/rpng/open_vins/commit/d171a03abc3ccb8641855acbc107a70d00a104ba. Not sure if there is an accuracy change, but it is better to match the dataset. Feel free to reopen if you still have concerns.

closed 2023-01-05 by goldbattle