rpng/open_vins #302: hard landing in uzh fpv dataset(indoor_forward_3)

ryanESX, 2023-01-22:
I tried to test  indoor_forward_3 uzh fpv dataset. I found this dataset has hard landing in the end of bag file cause odometry divergence. Does it has anyway to resolve this problem?

goldbattle, 2023-01-22:
I couldn't solve this besides stopping the bag early. I think the IMU
readings saturate at that point.

ryanESX, 2023-01-22:
The saturation of IMU.  Do you mean it exceed the maximum of acceleration in IMU? For example, I set the max acceleration of the IMU in +-4g. But the hard landing make it 10g or any number exceed +-4g.

goldbattle, 2023-01-24:
Yes, or that the higher accelerations are less sensitive.
I have not looked into this problem / dataset directly, so take my thoughts
with a grain of salt.

ryanESX, 2023-01-31:
Okay, I got it. I will keep looking another datasets.  I noticed uzh fpv dataset use Qualcomm Snapdrgon Flight. Their on-board IMU is commercial level. Euroc/RPGN dataset has better IMU(ADIS16448). Maybe it is another possible factor.

closed 2023-02-14 by goldbattle