rpng/open_vins #303: Roll and yaw have a constant offset compared to GroundTruth when running with EuRoC.

hungdche, 2023-01-23:
Hi, 

I'm trying to run OpenVINS with EuRoC, and it seems that although the xyz and pitch match well, both roll and yaw differ have a constant offset of about 50 degree compared to ground truth.   
[image]
[image]

The config file that I used is the [exact ones](https://github.com/rpng/open_vins/tree/master/config/euroc_mav) provided by OpenVINS. My code is [here](https://github.com/hungdche/open_vins/blob/master/ov_msckf/src/run_illixr_msckf.cpp) just in case. 

Any help for pointer is greatly appreciated. Thanks!

goldbattle, 2023-01-24:
This might be due to your use of the quaternion.
We use a JPL quaternion which is the rotation from the GLOBAL to the IMU
frame.
I recommend you try writing the quaternion x,y,z,w to file, and then
processing it.
When a JPL quaternion is written in this order, and read as a Halmiton
quaternion, it flips the rotation direction to IMU to GLOBAL.

hungdche, 2023-01-24:
Thank you for your response! It does in fact match now

closed 2023-01-24 by hungdche