rpng/open_vins #304: Foxy build fail

fratopa, 2023-01-29:
Hi I am running Ubuntu 20.04 and Ros2 Foxy. The build fails when trying to build ov_msckf. 
To solve the problem I h ad to change a line in the file located in the file `/open_vins_LDC/ov_msckf/src/ros/ROS2Visualizer.h`.
In particular I changed line 28 from:
`#include <image_transport/image_transport.h>`
to:
`#include <image_transport/image_transport.hpp>` 
I hope it helps anyone else with the same problem

goldbattle, 2023-02-14:
Seems that they changed the header fully? Originally the `.h` was kept in Galactic to support backwards compatibility with Dashing. It looks like they still have it on the github project:
https://github.com/ros-perception/image_common/tree/foxy/image_transport/include/image_transport

Can you see if you have the header file in your ROS install location?

If they actually removed the old `.h`  we now need to put some sort of a version check to make sure we include the correct one on older versions and the new `.hpp` on newer ones.

mzahana, 2023-06-23:
@fratopa Thanks a lot! I faced the same problem on ROS 2 `humble` and this fix worked for me.
I hope this issue gets fixed in future versions.

goldbattle, 2023-10-28:
It looks like the re-added the headers? I think to keep backwards compatibility I will keep it as is. It seems to build fine for me on humble and one ubuntu 22.04 with just the header warnings. If you have a way to reproduce on this please reopen.

closed 2023-10-28 by goldbattle