rpng/open_vins #305: Segmentation fault / freezing under heavy CPU load

pritzvac, 2023-02-09:
I'm running OpenVINS on ROS1 with stereo data from the RealSense T265 camera. I experienced sporadic crashing / freezing of the VIO under heavy CPU load.

According to GDB, sometimes a segmentation fault happens at https://github.com/rpng/open_vins/blob/98c55256a45b5518cf0a02e0f984cd2afafe99e0/ov_msckf/src/state/State.h#L67 (and sometimes the VIO freezes with one of the threads being in the same location). I assume some elements  are deleted from the `_clones_IMU` map by another thread while iterating over it.

Segmentation fault:
[image]

Backtrace:
[image]

I added mutex locks before the `for` cycle https://github.com/rpng/open_vins/blob/98c55256a45b5518cf0a02e0f984cd2afafe99e0/ov_msckf/src/state/State.h#L67 and before the marginalization and element removal in https://github.com/rpng/open_vins/blob/98c55256a45b5518cf0a02e0f984cd2afafe99e0/ov_msckf/src/state/StateHelper.cpp#L622 where I assume the elements are being deleted from the map. So far it seems to have fixed the issue (I'm testing it by running OpenVINS along with CPU and fileio benchmarks using `sysbench`). Should I submit a pull request?

goldbattle, 2023-02-14:
Yeah, if you are able to. I need to think a bit more about how to elegantly solve this, but I think your fix would solve things.

pritzvac, 2023-02-15:
Ok, I made the pull request. I tested it on some rosbag from the euroc dataset running along with the CPU benchmark as well and it seems to fix the issue.

PR #306 (2023-03-15): Fix: Segmentation fault in State.h
Fixes #305 by adding mutexes to State.h and StateHelper.cpp to avoid deleting elements while iterating over them.
closed 2023-03-15 by goldbattle