rpng/open_vins #308: Different Results When Testing The Same Dataset

SueMM00, 2023-02-20:
Hi, I'm trying to test ov_msckf by using my own dataset (recorded rosbag) 
When I test using the same dataset and same parameters, the trajectory shown on rviz can be quite different. Sometimes the trajectory is correct, but sometimes it drifts quite seriously.
May I know what could be the problem and is there a way to solve it?

goldbattle, 2023-02-20:
It is near impossible for me to help in anyway with your own sensors and this situation. My recommendation is to check that your calibration is of good quality, you have good feature tracking, you have calibrated your IMU noises and it is of reasonable quality. Maybe you are hitting some computational limit of your computer (which can cause divergence sometimes), but I am not sure.

SueMM00, 2023-02-21:
Thank you for your suggestion, it was due to incorrect calibration, and the problems is fixed now. Thank you very much again for making this great project! :)

closed 2023-02-21 by SueMM00