rpng/open_vins #310: Openvins stop working sometimes

EhrazImam, 2023-02-22:
Hi @goldbattle i am able to run it on my custom hardware and it runs perfect but after running it for several time on my hardware i can noticed 2/3 time it's stopped working. 
[image]

As you asked for the dataset over here https://github.com/rpng/open_vins/issues/297#issuecomment-1423181810 please find the link https://drive.google.com/file/d/1X3i3MtK4kXGncgBYEP02Q5zxWhZU7UwX/view?usp=share_link

EhrazImam, 2023-02-25:
@goldbattle Any update??

goldbattle, 2023-02-25:
No I have not had time, but thanks for posting the bag so I can take a look
when I get time.

EhrazImam, 2023-02-28:
Ok thankyou

EhrazImam, 2023-03-04:
Hi @goldbattle have you looked into this as today i crashed my drone while trying to do take off landing with the help of openvins which was running perfectly and after it stop working and drone got crashed.
Thankyou It will be great if you can have a look into this

goldbattle, 2023-03-06:
I assume this only happens during zero velocity update?
Is it if you do a zero velocity update for a *very long time*? or any
amount of time?
What type of platform are you running on (OS version, arm  / x86 etc..).

It looks like the chi2 is getting a NaN, which should only happen if the
covariance becomes NaN which is a very large problem (an EKF cannot work if
the covariance is invalid).

EhrazImam, 2023-03-06:
@goldbattle For zero velocity yes it do but not only it happens with zero velocity it happens with while velocity update.
I am Using Jetson 
How can i resolve chi2 error

goldbattle, 2023-03-15:
The bag file you have posted has zero timestamps for the image topics. Could you upload the bag that has the problem and has valid timestamps? Additionally the config files you are running which cause the problem.

EhrazImam, 2023-03-17:
Please find the calibration file that i am using [econ.zip](https://github.com/rpng/open_vins/files/11001712/econ.zip)
Please find the bag file link https://drive.google.com/file/d/1pqbX6whjo3egSdsm9bkyQ52Fd7V24rFE/view?usp=share_link

EhrazImam, 2023-03-21:
@goldbattle any update??

goldbattle, 2023-04-15:
I tested both stereo and mono with the bag / config you sent and was unabel to get it to NaN.
[image]

Could you try with the latest master?

EhrazImam, 2023-04-16:
@goldbattle Thankyou its been solved it was due to the stereo camera not its working fine.

goldbattle, 2023-04-16:
Yep. If you simulate with more cameras then it will use those 3 and 4 configurations.
```
roslaunch ov_msckf simulation.launch max_cameras:=3
```

Only a simulation thing since the ROS subscribers currently only support 1-2 cameras as input.
Glad you got things figured out.

closed 2023-04-16 by goldbattle