rpng/open_vins #312: NaN problem in dynamic initialization

CanCanZeng, 2023-03-04:
Hi, I got NaN velocity for sigmas_vel  and sigmas_bg and sigmas_ba sometimes in dynamic initialization, thus cause the system crash later. Maybe the calibration between camera and imu is not very accurate, can you give me some advice on solving this problem?

CanCanZeng, 2023-03-07:
Yes, you are right! I find that the NaN most probabaly caused by degenerate of motion. So I both increase the min reciprocal number and check the distance between the first frame and the last frame, if the distance is too small, the motion may degenerated. This strategy has worked so far, don't know if there will be other cases in the future.

goldbattle, 2023-03-07:
You can try to change the `init_imu_thresh` and `init_max_disparity` as when the platform is stationary static initialization can be used directly and skip dynamic initialization. If you have stereo, then you should be able to always initialize right away with the dynamic init (don't know if I have tested this, but since features can be triangulated, it should work).

closed 2023-03-07 by goldbattle