rpng/open_vins #326: Tracking with open-vins failing on Android

Madhuparna04, 2023-04-12:
I have been trying to use open-vins on an Android phone (Samsung Galaxy S22 Ultra).
I used Kalibr to do the calibration (20 HZ Camera images from one back camera and 200 HZ IMU).
But tracking is failing with the values I got from Kalibr.

[Kalibr Camera report](https://drive.google.com/file/d/1O8AlVgHRX6z9wegRgnt9QnL2cv83EQju/view?usp=share_link)
[Kalibr Cam-imu report](https://drive.google.com/file/d/1PumJGcWBC5Rc5uESVMjs9tH7h5kjfwM9/view?usp=sharing) 

I used [allan-variance](https://github.com/ori-drs/allan_variance_ros) for finding the gyroscope and accelerometer noise and random walk.

I am using monocular currently and the returned pose seems to be increasing constantly. Number of features found is either 1 or 0. Also, the chi2 value I am observing is really high (30000 to 50000).

Any suggestion on how I can get the correct calibration results so that tracking works with open-vins?
Thank you!

goldbattle, 2023-04-12:
Your transformation to the IMU to CAM in kalibr is over a meter.
I would get Kalibr to give a reasonable result before testing VIO.

goldbattle, 2023-04-12:
[image]

Madhuparna04, 2023-04-13:
I was able to get the right calibration values. Also for reference, [this](https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki) has some default values for Android phones. Thank you!

closed 2023-04-13 by Madhuparna04

Genozen, 2024-08-01:
@Madhuparna04 , sorry to bother. I'm kind of interested in running OpenVins in Android platform. Any chance you could shed some light on how to get it onto Android?