rpng/open_vins #327: Multicamera setup

EhrazImam, 2023-04-17:
Hi @goldbattle as you mentioned over here https://github.com/rpng/open_vins/issues/310#issuecomment-1510420888 can you help me with more details as i am interested to make it work with real hardware. Is it just supported in simulation.

goldbattle, 2023-04-17:
Yes this should likely work, but right now we consider that all cameras are synchronised (only a single time offset between the IMU and the camera set). If this is the case, then what you have should work. Additionally, you will likely need to edit the trackers to explicitly call either monocular tracking on each camera, or stereo tracking between any stereo pairs. I think things should work with more than 2 cameras if you set the `use_stereo` to `false`

https://github.com/rpng/open_vins/blob/74a63cf758f595cc280f2165e5a6576e7ef8521d/ov_core/src/track/TrackKLT.cpp#L78-L93

You could write something like the following:
```cpp
// track a stereo pair (first two in message) then the rest
if (num_images == 4) {
    feed_stereo(message, 0, 1);
    parallel_for_(cv::Range(2, (int)num_images), LambdaBody([&](const cv::Range &range) {
                    for (int i = range.start; i < range.end; i++) {
                      feed_monocular(message, i);
                    }
                  }));
}
// track each camera independently (multi-thread if enabled)
if (num_images == 4) {
    parallel_for_(cv::Range(0, (int)num_images), LambdaBody([&](const cv::Range &range) {
                    for (int i = range.start; i < range.end; i++) {
                      feed_monocular(message, i);
                    }
                  }));
}
```

EhrazImam, 2023-04-18:
@goldbattle thankyou for your reply.
The camera setup that i am using are all synchronized, where out of 1 stereo pair and 2 are mono. So i need to do more chages with this setup.

goldbattle, 2023-04-18:
If you just want to do mono tracking (independent), then you shouldn't need to. But to do a specific stereo pair, you will need to make edits similar to the ones I have suggested.

EhrazImam, 2023-04-19:
@goldbattle I did the changes in ROS1Visualizer.cpp and ROS1Visualizer.h but i am getting this issue

catkin_ws/src/open_vins/ov_msckf/src/ros/ROS1Visualizer.cpp:182:105:   required from here
/usr/include/boost/bind/bind.hpp:833:35: error: no match for call to ‘(boost::mfi::mf8<void, ov_msckf::ROS1Visualizer, const boost::shared_ptr<const sensor_msgs::Image<std::allocator<void> > >&, const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, int, int, int, int>) (ov_msckf::ROS1Visualizer*&, const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, int&, int&, int&, int&)’
  833 |         unwrapper<F>::unwrap(f, 0)(a[base_type::a1_], a[base_type::a2_], a[base_type::a3_], a[base_type::a4_], a[base_type::a5_], a[base_type::a6_], a[base_type::a7_], a[base_type::a8_], a[base_type::a9_]);
      |         ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

goldbattle, 2023-04-19:
I recommend googling. If I remember correctly, there is an upper limit to the number of parameters a function can take with the callback. So I suggest removing the `int` variables if you still have those.

closed 2023-05-09 by goldbattle