rpng/open_vins #329: Some questions about the simulator.

LarryDong, 2023-04-19:
Hi, thanks for sharing the codes, and the detailed documents. It really helps!

My aim is to generate trajectory GT, as well as IMU, maybe better with images for evaluating other VIO system. 
I've viewed the codes and online documents for some time but I found little tutorial about the simulation.

My questions are:
- It seems openvins simulator generate "virtual" features (as described in [sim-visbearing](https://docs.openvins.com/simulation.html#sim-visbearing), and used only in `feed_measurement_simulation`.  Is that right? If yes, how could it used for other VIO system (https://github.com/rpng/open_vins/issues/228#issuecomment-1040831761)? 
- Why simulation codes are under `ov_init` module? I found "test_simulation.cpp", and "sim/SimulatorInit" are under ov_init, which also needs `ov_core`'s lib, are these all the files I need for my simulation?
- How could I provide a trajectory? Is the trajectory file be `record_timing_filepath` in "config/rpng_sim/estimator_config.yaml", which should be the  simulation's "unset_path_to_config.yaml"? Do you provide a template for the trajectory file?
- I guess many people need a more detailed documents about the simulator, at least for me. [This page](https://docs.openvins.com/simulation.html) only explains the principle, not how to use the codes. Any more tutorials?

Again, thanks for this amazing repo.

goldbattle, 2023-04-19:
> Is that right? If yes, how could it used for other VIO system

You would need to implement this interface yourself, but from a high level it would involve replacing the other system's frontends with directly feeding pixel bearing with known matches.

> Why simulation codes are under ov_init module? 

There are two simulators. One for dynamic initialization and the other for the ov_msckf package. You can find both in the `src/sim/` folder in each package. Both are pretty much the same, so if you understand one it should be clear how the other one works.

> How could I provide a trajectory?

Please take a look at the code and dataset reader. There are a bunch of example files in the [ov_data folder](https://github.com/rpng/open_vins/tree/master/ov_data/sim).
If you simply inspect one of these files, you should be able to see what they contain:
```
# timestamp(s) tx ty tz qx qy qz qw
```

> I guess many people need a more detailed documents about the simulator, at least for me. [This page](https://docs.openvins.com/simulation.html) only explains the principle, not how to use the codes. Any more tutorials?

If you inspect the code there are plenty of comments and documentation.
Additionally each function has documentation.
Please take a look.
https://docs.openvins.com/classov__msckf_1_1Simulator.html

The simplest example of using the simulator should give you an example on how you could leverage it yourself:
https://github.com/rpng/open_vins/blob/master/ov_msckf/src/test_sim_meas.cpp

Hope this helps.

LarryDong, 2023-04-20:
Thanks for your reply. I would try those soon.

closed 2023-04-20 by LarryDong

956646930, 2025-02-16:
I want to know where the corresponding YAML file located when using the simulator