rpng/open_vins #339: question about the mapping backend

tadashi-hamada196, 2023-06-06:
I have read your paper on map-based visual-inertial localization and found it to be an excellent work! I was wondering if you have plans to add a mapping backend to open_vins in order to develop a drift-free slam system. I am aware of the ov_maplab repository in rpng, but it seems too complicted with several dependencies. Thank you.

goldbattle, 2023-06-08:
I think at this time we don't have any plans. The ov_maplab is for doing multi-session mapping, or offline optimization of the map. I do recommend using this if you really need recovery an accurate environmental map. We do have a demo called [ov_secondary](https://github.com/rpng/ov_secondary) which is a secondary optimization thread (last tested in Ubunutu 18.04, so might not work any more).

tadashi-hamada196, 2023-06-09:
get it，Thanks

closed 2023-06-09 by goldbattle